indilib/indi #1033: myFocuserPro2 driver GUI issue

CheekyMonkey, 2019-10-19:
@ajt68 Thanks for writing the MyFocuserPro2 indi driver

Gave version 4 of the MyFocuserPro2 indi driver a sanity test run this week

Not the best night for it with the clouds rolling in, but gave it a shot

Video below of a focuser test on Caph.
Clouds made a mess of the V curve, but otherwise a good outcome

https://youtu.be/T186YPI6_Gk

One observation, not confined to version 4 of the MyFP2 indi driver, but seems to be an issue on versions 1,2,3 and 4:

The focuser **"Absolute Position"** and **"Maximum Position"** fields do not always show the current values both on the focuser settings page and also on the indi focuser page

See highlighted fields in red in the screen captures below for reference

Seems somewhat inconsistent as the two values do sometimes persist after a closing then re-opening the indi/ekos windows but the values do not always seem to survive a full shutdown/restart of indi

**Absolute Position and Maximum Position not reflecting in the GUI**
[image]

**Absolute Position and Maximum Position not reflecting in the GUI**
[image]

**Focus on Caph**
[image]

**MyFocuserPro2 firmware version 291**

knro, 2019-10-20:
So this is not an INDI issue (Ekos is just one client of INDI out of many). But at any rate, in INDI control panel in Ekos, the value on the right is the target, so it could be any number. What matters if the current value on the left.

closed 2019-10-20 by knro

CheekyMonkey, 2019-10-20:
Thanks @knro 

That's good to know

I'm calling it out as an issue as the fields highlighted in my screen captures in red are sometimes populated and sometimes not which seems odd

Shutting down the client windows and opening them up again will sometimes **not** show a value in the field then for the same session shutting down the client windows and stopping/re-starting indi will **sometimes** show a value in the field

I've not yet worked out the exact pattern/scenario to reproduce this consistently but I'll stick with it until I do. 

I observe that the same occurs with the kstars/indi Windows build (Windows 10) as well as the kstars/indi build on Stellarmate on the Raspberry Pi 

I'll see if I can capture the behaviour on video then post here

CheekyMonkey, 2019-10-20:
I'm now able to reproduce the issue consistently

Steps to reproduce the issue:

I'll walk through the steps to reproduce the issue based on Stellarmate as that's the build I use

1. Start up the Stellarmate Web Manager
stellarmate.local:8624

2. Start the INDI server
All devices defined in the equipment profile will start-up as expected

3. Open Kstars
4. From the Kstars "Tools" menu select "Ekos"
5. "Start" Ekos from the "Ekos Profile - KStars" GUI
In my case, I have it set in my equipmemt profile to "auto connect" all devices

6. Click "Connect" to connect all devices defined in the equipment profile
7. Once the "INDI Control Panel - Kstars" screen opens then navigate to the MyFocuserPro2 page

Note the values shown in "Absolute Position" and "Max Position" fields
e.g. In my case see the first screen capture below

Note the empty fields for Absolute Position and Max Position

8. Close the "INDI Control Panel - Kstars" screen
9. Click "Disconnect" to disconnect all devices
10. Stop Ekos
11. Close Kstars

Follow steps 3 through 7 again
Now note that this time once you get to step 7, the Absolute Position and Max Position fields are now populated as expected
e.g. In my case see the second screen capture below

12. Close all screens and disconnect devices
13. Stop the indi server via the Stellarmate web manager
stellarmate.local:8624

Repeat the steps 1 through 7 to reproduce the issue again

If you don't use Stellarmate, substitute steps 1,2 and 13 for the equivalent steps for your set-up

[image]
[image]
[image]
[image]