indilib/indi #1034: Liveview FIX for Sony Alpha Cameras

v4mp1, 2019-10-19:
https://indilib.org/forum/ccds-dslrs/5881-sony-alpha-5100-viewfinder-widget-error-fixed.html

knro, 2019-10-20:
Thank you for the fix. Is this applicable to all Sony cameras?

v4mp1, 2019-10-20:
I have only the 5100 but my guess is it works for the whole Sony DSLM android based firmware lineup.

knro, 2019-10-21:
Ok I applied this to master, can you test and verify?

v4mp1, 2019-10-21:
I don't see any changes there? https://github.com/indilib/indi-3rdparty/tree/master/indi-gphoto

knro, 2019-10-21:
You're right, I pushed it to the gphoto_autoexposure branch by mistake. Can you test it there?

v4mp1, 2019-10-21:
Alright, I can test it in some hours. Thanks.

v4mp1, 2019-10-21:
Jasem it works but you forgot to add Sony in the stop preview section.
I have to click twice on the X to close the preview window:

int gphoto_stop_preview(gphoto_driver *gphoto)
{
    // Olympus cameras support streaming but without viewfinder_widget
    if (strcasestr(gphoto->manufacturer, "OLYMPUS"))
        return GP_OK;

v4mp1, 2019-10-22:
Everything works now Jasem you can upload to master.

Thanks.

closed 2019-10-22 by v4mp1