indilib/indi #1045: Build: 2019-11-20T07:23:51Z / Ekos crashes with indi-sx-x

escribana, 2019-11-23:
**Describe the bug**
A clear and concise description of what the bug is.  Provide a general summary of the issue in the Title above as well as affected program name like `indiserver` or `indi_toupcam_ccd`

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Perform action
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:
 - Version [e.g. 1.7.6, GIT..etc]

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

knro, 2019-11-24:
Ok, so this is technically a KStars issue, not INDI. However, do you have a backtrace for the Ekos crash? run **gdb -ex run --args kstars**

then when it crashes, type in **backtrace** to get the backtrace and share it.

escribana, 2019-11-24:
OK! Thanks. I'll do this. But where is the corresponding bug report site for KStars?

knro, 2019-11-25:
bugs.kde.org

closed 2019-11-25 by knro