indilib/indi #1056: ASI ZWO 174 GAIN always at max when in 16 bit RAW

fiveseve84, 2019-12-14:
Hi I am using an ASI ZWO 174MM-Cool via USB 3 on an Raspberry Pi 4 4GB model. The camera is powered.

Everthing works fine so far, BUT when setting to 16bit RAW GAIN is always at 400 (maximum). Using 8 bit mode gaing works as expected. To be exact the GAIN settings does not work and is being ignored. The resulting image is always at max GAIN (same as 8bit and GAIN set 400).

Raspbian Buster
4.19.75-v7l+

Indi installtion as in:
https://www.indilib.org/download/raspberry-pi.html

Client application is Kstars 3.3.8

knro, 2019-12-15:
How about when you set it to say, 50 in Ekos? Does it change as well when you capture in INDI control panel?

fiveseve84, 2019-12-15:
Sorry forgot to mention. I tried that already. 50 120 240 all look like Max gain. And it does not make any difference changing the value in Indi panel or in cam settings.

Also tried a lot different settings to see if behaviour changes. Auto gain on off, Max auto gain exposure etc . But still always at Max gain.

knro, 2019-12-15:
Have you tested with ASICAP? same results or it works there?

fiveseve84, 2019-12-15:
Not asicap specifically. But imaging with the setup since nearly two years. Was using sharpcap with Windows before. (on an Intel nice mounted to the telescope) There everything works fine. 

When I saw I can ommit windows rdp and use a less power hungry Rpi. I was searching for exactly your solution but never found something. Then I saw an Indi/Kstars video on YouTube and was immediately sold.

But let me try sharpcap again. Maybe there is an  issue with the I am not aware of, since I didn't use it  since last winter

fiveseve84, 2019-12-15:
Okay somehow very interesting. Sharpcap works as expected. 
In addition installed Ubuntu 
Description:	Ubuntu 18.04.3 LTS
4.15.0-72-generic
On another Intel NUC @ CPU J3455
As described:
https://indilib.org/download/ubuntu.html
apt-get install indi-full gsc

And again it works as expected. 

So the issue hast smething to do with either the PI4, OS or a difference between packages RPI4/X86_64

One another thing I can see, the grain or image output grain behavior looks completeley different between RPI and NUC

Intel NUC:
[image]

RPI4
[image]

fiveseve84, 2019-12-15:
Maybe an additional note: if exposure settings are orrect, I can see a picture with the RPI4. So the image or readout is not broken. Only GAIN/grain behaviour very strange

knro, 2019-12-15:
If you saved the FITS image from RPI4 and opened it on another PC or NUC, does it show the same pattern?

fiveseve84, 2019-12-15:
Found the issue. Sorry for the buzz. Apparently Kstars has an independent Ekos settings where it auto stetches all images.  But this is only applied to 16bit images and althoug while capturing I deaktivated Stretchin, Contrast etc.

Sorry. Issue can be closed

closed 2019-12-15 by knro