indilib/indi #1059: QHYCCD drivers [ERROR] GetQHYCCDSingleFrame error (-1)

AradoSKY, 2019-12-26:
Hello,

Here are two files showing QHYCCD drivers for PoleMaster and QHY5LII are not playing well with INDI. These two drivers error out with [ERROR] GetQHYCCDSingleFrame error (-1). This prevents polar alignment for KSTARS. On Random occasions, Polar Alignment with QHYCCD will work. QHYCCD drivers work after multiple [ERROR] GetQHYCCDSingleFrame error (-1) errors. No rhyme or reason is given in any INDI logs.  

Attempting different USB settings, buffer sizes, resolutions, capture seconds, and others does nothing to prevent [ERROR] GetQHYCCDSingleFrame error (-1).  I have another camera, ZWO ASI 120MM. The ASI 120MM works as described. 

When capturing, under Polar Align, QHYCCD drivers fail after 67 seconds. EKOS shows the QHY disconnect with INDI. 

All QHYCCD drivers enumerate without issue within INDI. When attempting to use, the QHY drivers dump many [ERROR] GetQHYCCDSingleFrame error (-1) and mostly never work.   
 

[2019-12-26-PolarAlign-Errors.txt](https://github.com/indilib/indi/files/4003410/2019-12-26-PolarAlign-Errors.txt)

[2019-12-24 PolarAlign Camera.txt](https://github.com/indilib/indi/files/4003411/2019-12-24.PolarAlign.Camera.txt)

knro, 2019-12-27:
Please enable debug logging and re-attach the logs.

AradoSKY, 2019-12-27:
Hello Knro,

Thank you for checking. Here are my recent efforts of debugging. I am using the PoleMaster for Polar Alignment. I get the same errors with the QHY5L-II-C. KSTARS crashes now when attempting to Polar Align with PoleMaster and EKOS. 

[indi_qhy_ccd_13:41:32.log](https://github.com/indilib/indi/files/4005442/indi_qhy_ccd_13.41.32.log)

AradoSKY, 2019-12-29:
Hello,
The INDI nightly arrived. I update the 85-qhyccd.rules file with your supplied file from another thread. I have updated to the current release. I conducted a test with the PoleMaster and QHY5L-II-C. Both continue to fail with [ERROR] GetQHYCCDSingleFrame error (-1). Both LOG files are too big to upload. I will attempt to compress and try again. 
[log_20-20-00.txt.tar.gz](https://github.com/indilib/indi/files/4007675/log_20-20-00.txt.tar.gz)
[indi_qhy_ccd_01:20:36.log.tar.gz](https://github.com/indilib/indi/files/4007676/indi_qhy_ccd_01.20.36.log.tar.gz)

knro, 2019-12-29:
Ok I was wondering if this is due to updated libqhy? can you build the indi-qhy driver against older libqhy from Github and see if that works?

AradoSKY, 2019-12-29:
Hello Jasem,

Thank you for taking this problem. Yes. I will seek to remove libqhy and indi-QHY. I will then install indi-qhy and check. If ok, then I will see about installing libqhy.

AradoSKY, 2019-12-30:
Hello,
I think I found the problem.  I completed the removal of all things QHY. Then I tested. This morning on a whim, I decided to check the SYSLOG.  This is what I found during the testing. 

Dec 30 09:00:43 astroberry kernel: [ 7592.169498] usb 1-1.2.4.1: usbfs: process 4686 (indi_qhy_ccd) did not claim interface 0 before use

This occurs whenever the Polar Align begins a capture. The capture dies and Solver aborts.

knro, 2019-12-30:
ONLY when you do polar alignment? how about regular capture? looping..etc? There is nothing special about polar alignment per se.

AradoSKY, 2019-12-30:
Hello, 

I just tried to conduct a stream. Yes. Even with a stream, this error with indi_qhy_ccd shows up in SYSLOG.

AradoSKY, 2019-12-30:
=
]\

AradoSKY, 2019-12-30:
Hello,

Another option is not issue a claim against the indi_qhy_ccd driver. Look at  tristil commented on May 20, 2014. 

https://github.com/tessel/node-usb/issues/30

knro, 2019-12-30:
That's fine. However, all the low-level USB communication is happening inside the QHY SDK library and not the INDI driver. So QHY needs to fix these issues inside the SDK.

AradoSKY, 2019-12-30:
Hello,
Thank you. This explains the problem then. Is an issue going to raised to QHY?

AradoSKY, 2019-12-30:
Hello,
I have confirmed. If I take one camera completely out of the EKOS set, each camera works as designed. Having them together, the indi_qhy_ccd driver does not handle more than one camera from a vendor at a time.  

This means the ZWO and PoleMaster are it. I would like to use the QHY5LII-C. Bummer.

closed 2019-12-30 by AradoSKY