indilib/indi #1061: [ERROR] Exposure failed after 3 attempts with a ASI533 camera

mg95-16, 2020-01-07:
I'm using an ASUS Thinker Board Ubuntu 18.04.3 LTS
I have the latest version of indi-full.
I created a profile with a ZWO CCD camera.
When I start this profile everything is OK.
But when I want to take a photo, I get the following error.

- "Capturing 1,000-second  image..."
- ZWO CCD ASI533MC Pro :  "[ERROR] Exposure failed after 3 attempts. "
- "Capture failed. Check INDI Control Panel for details."
- "Restarting capture attempt #1"
[log_22-57-49.txt](https://github.com/indilib/indi/files/4032846/log_22-57-49.txt)

knro, 2020-01-08:
It *might* have to do with Tinkerboard USB 3.0 port. Did you try on RPi3/RPi4? on USB 2.0 port?

mg95-16, 2020-01-08:
No I only have a Tinker Board so only USB 2.0

mg95-16, 2020-01-23:
It s OK with RPi4 on USB 3.0 port

closed 2020-01-23 by mg95-16