indilib/indi #1078: No bit depth switching using indi_toupcam_ccd

raul-chirea, 2020-01-25:
**Describe the bug**
indi_toupcam_ccd driver does not offer the possibility to switch between bit depths, although the hardware of the camera supports it. For example, I have a Touptek GP2000KPA camera. Its specification says that camera supports bit depths of 8bit and 12bit but, with the current driver, it can not be used in 12bit mode (only 8bit mode).

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. In the "Controls" tab of the driver control there is no way to switch from 8bit mode to 12bit mode as described in documentation (https://www.indilib.org/devices/cameras/toupcam-cameras.html - "Capture" section).

**Expected behavior**
indi_toupcam_ccd should be able to use 12bit mode of the camera.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linux (Fedora 28) + driver running on RPi 2
 - INDI Version: 1.8.2

knro, 2020-01-25:
RGB is 8bit. Try RAW and see what can you get.

raul-chirea, 2020-01-27:
Wow. Magic. Changing from RGB to RAW switches camera to 12bit. A little bit unintuitive, but it works. Thanks !

raul-chirea, 2020-01-27:
Somebody tell me, please, should I close the bug or let it open (maybe the docs should be modified to be more explicit) ? I don't know the customs around here.

knro, 2020-01-28:
Yes, I updated the driver documentation to help users know about this.

closed 2020-01-28 by knro

raul-chirea, 2020-01-30:
Excelent ! Thanks a lot, again. BTW, great job with EKOS.