indilib/indi #1085: ML16000 all frames bad, frame output vary substantially

gschwim, 2020-02-03:
**Description**
I'm consistently seeing very poor readouts from the ML16000 using indi & Ekos. For example, the following  60 second light frame which appears to be missing a lot of the light signal and also has an odd artifact at the upper left. This artifact changes and is not always present in a frame:
[https://www.dropbox.com/s/8r62t7gvnbu6d3u/M_42_Light_60_secs_2020-02-02T21-57-07_001.fits?dl=0](url)

A subsequent light captured using TheSkyX (TSX) **without power cycling** the camera gives a similar result. Note the artifact has disappeared. This happens with both systems.
[https://www.dropbox.com/s/e9w0t12lhfnytxl/00000003.fit?dl=0](url)

**Power cycling the camera** and using TSX yields consistently correct outcomes:
[https://www.dropbox.com/s/d7xbbsguhwrnri8/00000006.fit?dl=0](url)

Bias and dark frames are similar. This bias frame has a median of ~41K(DN):
[https://www.dropbox.com/s/a8plnayug4u1bpm/Bias_0.040_secs_2020-02-02T20-12-31_001.fits?dl=0](url)

This dark frame has an odd pattern in it:
[https://www.dropbox.com/s/ow9hgpf6c3lhr1g/M_42_Dark_60_secs_2020-02-02T21-55-21_003.fits?dl=0](url)

Whereas after a power cycle TSX will yield a proper result consistently. Bias:
[https://www.dropbox.com/s/oju4dkechdpg7tx/00000008.Bias.fit?dl=0](url)

**To Reproduce**
This is easy to reproduce:
1. Power the camera on
2. Start Ekos, take some frames - all will be bad
3. Close Ekos, start TSX - all frames will be bad
4. Power cycle the camera - TSX frames all good

**Expected behavior**
Indi should produce correct frames consistently.

**System Configuration**
Running Linux Mint 19.3 w/ PPA packages installed, plus freshly built Kstars as of today. Also saw the same behavior running indiserver in VirtualBox Ubuntu Server guest on a Windows host with Kstars running on Windows.

The FLI ccd driver is at defaults. Same behavior in forcing 4Mhz mode.

**Log Files**
To be gathered...

gschwim, 2020-02-03:
Tested with an ML16200 (KAF-16200 sensor) camera, does not produce these symptoms. Problem seems to be specific to the ML16000 (KAI-16000).

knro, 2020-02-04:
I asked FLI to look into this. You can communicate with them directly as well to raise an issue since we rely on their SDK to communicate with the camera.

gschwim, 2020-02-04:
Will do. Same as last time I guess.

knro, 2020-02-24:
This was fixed by disabling background flushing. Closing.

closed 2020-02-24 by knro