indilib/indi #1088: Astronomical calculator used for simulation and previsions

iliaplatone, 2020-02-07:
There are some unused features that need explanation and testing in a fake or virtual environment.
Some of them may be alignment, photon flux calculation, magnitude, size and distance estimators and various common functions like Celestial to local coordinates, which can be like to be written into a driver that returns their results in a text/property value for external/client computation.

The proposal is to write a new driver (simulator like) that includes various algorithms and calculations useful in astronomy. Each branch of algorithms or functions may be part of a tab or group that is inherited with the calculation usefulness (for example visual conversions group should have FOV calculator etc., Astrophysical constants ... etcetera).

Most of these functions are already present into KStars or the various clients, this proposal is intended for developers' tests or simple clients or loggers.

I personally wrote some code that won't have immediate implementation but need test and feasibility/reliability lookup.

knro, 2020-02-07:
I can see how this can be useful... however the primary purpose of INDI drivers is to communicate with hardware, or facilitate communication with hardware (i.e. bridge). What you propose clear belongs in the client land. We can have tools (i.e. scripts) that  perform that and provide such functionality for testing?

iliaplatone, 2020-02-10:
The target of the proposal is to test internal INDI functions for debugging, (indicom functions for example) for usage in future drivers. I was thinking about a all-function device simulator (while writing I wondered), nothing else.

closed 2021-09-28 by knro