indilib/indi #1089: 1.8.4 breaks build in websocket

darix, 2020-02-08:
```
[ 25%] Building CXX object CMakeFiles/indidriver.dir/libs/indibase/indigps.cpp.o
/usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DGIT_TAG_STRING=\"1.8.4-tgz\" -DHAVE_LIBNOVA -DHAVE_WEBSOCKET -DWITH_ENCLEN -DWITH_MINMAX -Dindidriver_EXPORTS -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/build -I/home/abuild/rpmbuild/BUILD/indi-1.8.4 -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/build/indidriver_autogen/include -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/indibase -I/usr/include/cfitsio -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/dsp -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/stream -I/usr/include/libnova -I/usr/include/libusb-1.0 -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/webcam  -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type  -g -DNDEBUG -std=c++11 -D_FORTIFY_SOURCE=2 -fstack-protector-all -fPIE -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-unused-but-set-variable -Wno-format-truncation -g -O2 -g -DNDEBUG -fPIC   -fPIC -o CMakeFiles/indidriver.dir/libs/indibase/indigps.cpp.o -c /home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/indibase/indigps.cpp
/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/indibase/dsp/dspinterface.cpp: In member function 'bool DSP::Interface::uploadFile(const void*, size_t, bool, bool, const char*)':
/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/indibase/dsp/dspinterface.cpp:519:13: error: 'HasWebSocket' was not declared in this scope
  519 |         if (HasWebSocket() && WebSocketS[WEBSOCKET_ENABLED].s == ISS_ON)
      |             ^~~~~~~~~~~~
[ 25%] Building CXX object CMakeFiles/indidriver.dir/libs/indibase/indiweather.cpp.o
/usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DGIT_TAG_STRING=\"1.8.4-tgz\" -DHAVE_LIBNOVA -DHAVE_WEBSOCKET -DWITH_ENCLEN -DWITH_MINMAX -Dindidriver_EXPORTS -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/build -I/home/abuild/rpmbuild/BUILD/indi-1.8.4 -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/build/indidriver_autogen/include -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/indibase -I/usr/include/cfitsio -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/dsp -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/stream -I/usr/include/libnova -I/usr/include/libusb-1.0 -I/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/webcam  -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type  -g -DNDEBUG -std=c++11 -D_FORTIFY_SOURCE=2 -fstack-protector-all -fPIE -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-unused-but-set-variable -Wno-format-truncation -g -O2 -g -DNDEBUG -fPIC   -fPIC -o CMakeFiles/indidriver.dir/libs/indibase/indiweather.cpp.o -c /home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/indibase/indiweather.cpp
/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/indibase/dsp/dspinterface.cpp:519:31: error: 'WebSocketS' was not declared in this scope
  519 |         if (HasWebSocket() && WebSocketS[WEBSOCKET_ENABLED].s == ISS_ON)
      |                               ^~~~~~~~~~
/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/indibase/dsp/dspinterface.cpp:519:42: error: 'WEBSOCKET_ENABLED' was not declared in this scope
  519 |         if (HasWebSocket() && WebSocketS[WEBSOCKET_ENABLED].s == ISS_ON)
      |                                          ^~~~~~~~~~~~~~~~~
/home/abuild/rpmbuild/BUILD/indi-1.8.4/libs/indibase/dsp/dspinterface.cpp:524:13: error: 'wsServer' was not declared in this scope; did you mean 'server'?
  524 |             wsServer.send_text(std::string(FitsB.format));
      |             ^~~~~~~~
      |             server
```

1.8.3 still builds.

also there are a lot of compiler warnings  that can be seen here https://build.opensuse.org/public/build/home:darix:playground/openSUSE_Tumbleweed/x86_64/indi/_log

knro, 2020-02-24:
Please test again against master.

darix, 2020-02-24:
master works better. but there is still alot warnings from -Wstringop-truncation (which is now one of our openSUSE default warnings) most are about destination and source having the same length. but some seem also seem overruns.

closed 2020-02-25 by knro