indilib/indi #1090: indi 1.8.x fails to build on armv7 arch

david-geiger, 2020-02-10:
Hi,

On Mageia Cauldron (I'm in the Mageia packager team) I tried to compile latest 1.8.4 release but it fails to build only on armv7 arch with:

> /home/iurt/rpmbuild/BUILD/indi-1.8.4/libs/indibase/inditelescope.cpp: In member function 'bool INDI::Telescope::processTimeInfo(const char*, const char*)':
/home/iurt/rpmbuild/BUILD/indi-1.8.4/libs/indibase/inditelescope.cpp:1699:17: error: 'stime' was not declared in this scope; did you mean 'ctime'?
 1699 |                 stime(&raw_time);
      |                 ^~~~~
      |                 ctime
make[2]: *** [CMakeFiles/indidriverstatic.dir/build.make:417: CMakeFiles/indidriverstatic.dir/libs/indibase/inditelescope.cpp.o] Error 1
make[2]: Leaving directory '/home/iurt/rpmbuild/BUILD/indi-1.8.4/build'
make[2]: *** Waiting for unfinished jobs....

I see that fedora also have the same issue with release 1.8.1 on armv7

Regards,
David

pterjan, 2020-02-10:
That code exists only on arm: https://github.com/indilib/indi/blob/master/libs/indibase/inditelescope.cpp#L1688

stime is part of time.h for C but doesn't appear to be in ctime for C++.

I have no idea why the code only tries to set the time on ARM.

knro, 2020-02-10:
It's there to correct the time on embedded device which are mostly arm. What needs to be done to fix the compile error?

pterjan, 2020-02-10:
This is a change in glibc https://lwn.net/Articles/811275/

> The obsolete function stime is no longer available to newly linked
>  binaries, and its declaration has been removed from <time.h>.
>  Programs that set the system time should use clock_settime instead.

commit 0cc0e242 (2020-02-24): Disable stime on arm
Fixes #1090
closed 2020-02-24 by knro