indilib/indi #1091: Errors in Ekos configuration options for INDI and/or indi_gpsd

Avocette99, 2020-02-11:
There are unexpected interactions when setting options for GPS in both the Ekos configuration for INDI where you may select 'GPS updates KStars' and Time and/or Location, and when running an Ekos equipment profile which includes GPSD, in the obtaining of GPS Fixes and the use which is made of them. 
The first image shows the selection options in Ekos options.
[image]
When Ekos is started and we look at the GPSD Option Tab of the INDI Control Panel and load the Default Configuration, the GPS button is highlighted and we presume that GPS data will update the KStars time and geographic location. However the KStars time increments by only one second for every GPS Fix obtained. So that if we set the timer for 60 seconds, KStars time advances for 60 seconds but is then reset by effectively 59 seconds since the UTC time shown in the GPSD Main Control Tab is only incrementing by a second with each fix. The Structure of the UTC time display is shown here.
[image]
If now in the Configuration options, the System button is selected as shown:
[image]
Then the format of the UTC time changes and the KStars time increments in line with real time as expected. If KStars time is modified manually the Set Time to Now follows real time.
[image]
You may note that although the GPS is set to refresh every 60 seconds, the Status box shows two GPS fixes each minute (at 13 and 17 seconds past each minute). This seems quite random - on some occasions there are apparently three GPS fixes per 60 seconds, possibly with two given the same time in the Status display.

I am running KStars 3.4.0 on a Raspberry Pi 4 4GB under Raspbian 3.2 (January 2020) and indi_gpsd version 0.5

I am keen to use the USB GPS dongle to set up my astrophotography server RPi controller in Real Time and physical Geographical Location since I use it in the field in HotSpot mode away from the Internet.

Many thanks

knro, 2020-02-11:
So GPSD driver does not return accurate time when you press `Refresh`?

Avocette99, 2020-02-11:
In (my) simple terms, in the second image which applies when the INDI Control Panel GPSD Configuration Options is set to GPS as the Time Source, then the time shown as UTC Time increments from the KStars and/or System time by one second per GPS Fix request, but not with the GPS receiver real time (e.g.tested with gpsmon). Pressing the GPS refresh button has the same effect, sometimes incrementing the UTC Time as a sort of counter by only one second but with no relationship with GPS receiver data. When System is selected as the time source, then GPS Fixes are correctly shown in the UTC Time display, but not used to update KStars time (nor System time). The geographic location is updated by a GPS Fix wtih whatever setting is made in the Configuration Options.

knro, 2020-08-30:
This was fixed recently in GIT.

closed 2020-08-30 by knro