indilib/indi #1102: Commit Use INDI_ENABLED and INDI_DISABLED as universal common names for On… has Sideeffemts on Indi 3rd-Party

dirkschwarze, 2020-03-14:
**Describe the bug**
Linking Indi 3rd Party fails with: indi-3rdparty/indi-asi/asi_focuser.cpp:375:19: error: ‘REVERSED_ENABLED’ was not declared in this scope
  375 |     FocusReverseS[REVERSED_ENABLED].s  = reversed ? ISS_ON : ISS_OFF;

**To Reproduce**
Try to Build Indi 3rd Party with current master branch...

**Expected behavior**
Commit has to be ported to Indi 3rd Party

TallFurryMan, 2020-03-14:
Could you confirm indi and indi-3rdparty are matching, in terms of code source and installed libraries/headers?

knro, 2020-03-14:
It was already updated in INDI-3rdparty 8027fb5f66e7ea266fecf744203ee322a9dda6e1

closed 2020-03-14 by knro

dirkschwarze, 2020-03-14:
Wierd, now I rechecked and saw Jasems commit from yesterday, sorry ....

Regards

Dirk