indilib/indi #1107: Question on pier side - inditelescope.cpp

fdesvallees, 2020-03-26:
When testing Ekos meridian flip on my TeenAstro driver, I found that it did not work as expected: 

Expected behaviour:
Ekos schedules a meridian flip when telescope is on the **west** side of the pier, pointing **east**. (as reported on the mount tab of the INDI control panel), and is about to cross the meridian (plus the given offset) 

Instead it schedules the flip when telescope is on the east side pointing west. Obviously it does not work since the mount is already past the meridian.

I switched the reporting of the pier side in my driver. (when telescope is on the East side I call  setPierSide(INDI::Telescope::PIER_WEST)
Now it works, but the reporting of the pier side in the INDI control panel (which is done in inditelescope.cpp) is wrong . I could override this function in my driver, but I would like to ask advice.
Apparently someone else has seen this issue, since the Synscan driver has this comment: (line 429)
`
  // Pier side
    if (m_isAltAz == false && sendCommand("p", res))
    {
        m_MountInfo[MI_POINT_STATUS] = res[0];
        // INDI and mount pier sides are opposite to each other
        setPierSide(res[0] == 'W' ? PIER_EAST : PIER_WEST);
    }
`
Thanks

knro, 2020-03-27:
@Chris-Rowland , @sterne-jaeger any thoughts?

Chris-Rowland, 2020-03-27:
Mounts vary as to how they report pier side and some have to reverse the value reported by the INDI driver to be consistent.  I have to do this with the Celestron mounts.  I don't recall any problem with the reporting of pier side in the mount driver.

I would expect the mount driver to report the pier side using the INDI standard.

Just trying it:
Mount AVX, using Celestron GPS driver. Flip set to 2 deg past meridian.
Slewed to Algenib, just East of meridian. 
Driver reports Pier West (looking East).
EKOS reports 9 minutes to flip.
Mount looks correct.
Waiting 6, 4, 3, 2, 1, 0
Flip started at HA +8 mins
Mount reports East (looking West)

All seems fine.

The last time I looked EKOS doesn't use the pier side property to manage the meridian flip.  I think it should but that's a different issue.

Chris

sterne-jaeger, 2020-03-27:
Same situation for example with the Avalon StarGO Firmware, which answers the opposite as it would be expected from a "pier side" logic. Here the code snippet:
```
    switch (answer)
    {
        case 'X':
            LOG_DEBUG("Detected pier side unknown.");
            setPierSide(INDI::Telescope::PIER_UNKNOWN);
            break;
        case 'W':
            LOG_DEBUG("Detected pier side west.");
            setPierSide(INDI::Telescope::PIER_EAST);
            break;
        case 'E':
            LOG_DEBUG("Detected pier side east.");
            setPierSide(INDI::Telescope::PIER_WEST);
            break;
        default:
            break;
```
And as you might see from the log output, I was confused as well and the log writing statement are mixed up.

As Cris rightfully stated, it varies by the mounts.

HTH
Wolfgang

Chris-Rowland, 2020-03-30:
Is this a bug in EKOS?  Other drivers seem to report and implement pier side correctly.

sterne-jaeger, 2020-03-30:
No, it's the diversity we see with the mounts.

Chris-Rowland, 2020-03-30:
I don't understand what you are saying.  If a GEM is looking at an object that is rising in the East then the INDI pier side will be West (looking East) for all mounts.  That is how the pier side is defined in INDI.

The mount tracks to the meridian and past it and the PierSide remains West, even though the mount is now looking a little West of the meridian.  The pier side does not change until a slew is done.

After the mount has tracked through the meridian then a slew to the current coordinates shoud do a long slew through the pole, ending up with the same coordinates but the scope now on the other side of the meridian. the PierSide has changed to East (looking West).

The mount can implement PierSide however it likes internally but if a mount doesn't report it as defined in the INDI interface then I don't see how it can comply with the INDI specification.  A consistent interface is essental if an application is expected to work with multiple mount types.

Chris

sterne-jaeger, 2020-03-30:
I just checked it with my Avalon. The pier side is shown correctly in EKOS, so when my mount is on the east side pointing west, the INDI driver shows "East (pointing west)" - so everything is fine.

Nevertheless, the Avalon mount firmware answers 'W' when the mount is on the east side of the pier. Slightly strange, but that's what I meant with variety of mounts. It seems like TeenAstro has a similar behavior as the Avalon mount.

But that's not a problem, as long as the one implementing the driver understands, that 'W' means "pointing west".

Chris-Rowland, 2020-03-30:
We seems to be saying the same thing in different ways.  The pier side that is reported in INDI must be the same for all mounts.

The Celestron mounts are the same, the report they direction the mount is looking before it has crossed the meridian so I have to change 'E' to PierWest and 'W' to PierEast to follow what INDI requires.

Maybe the OP's mount driver is not doing this, or not as required in some way.

Can we say that this is NOT a bug in EKOS.

fdesvallees, 2020-03-30:
Thanks to all, this is very useful. I still don't fully understand the issue, but here is my summary:

- From a quick check, Avalon, Celestron, SynScan and TeenAstro report the opposite side. All other mounts seem to report normally.
But I don't know which mount users have tested if the meridian flip works for them.

- As far as I can tell from the code, Ekos does not check the pier side when planning the flip. It only checks the initial HA of the object after a slew. If negative (E of meridian), it plans the flip. If positive it does not. Clearly this can cause problems, since TeenAstro (and probably others) can perfectly slew W of meridian while keeping the telescope on the west side, (with the counterweight bar above the telescope) as long as its limit is not reached, and therefore will never request the flip. 

This still does not explain why I have to report the opposite pier side in order to get a flip!
So I will keep looking into this, and let you know what I find. I would like to have the advice of the person who wrote the Ekos code (is it you, Jasem?)

Thanks!
François

sterne-jaeger, 2020-03-30:
Hi François,
I'm the one who refactored the meridian flip functionality around a year ago, so I know it quite well. I do not have any explanation why reporting a pier side has influence on the meridian flip. As you rightfully stated, it only depends upon the initial HA, which is negative east of the meridian. As soon as the current HA turns positive (plus a configurable delay), the EKOS mount controller re-issues a slew to the current position.

The other way round: it does not matter, whether the mount is able to slew beyond the meridian. Simply change the delay to that amount that is within your limits and the flip should happen smoothly.

- Wolfgang

sterne-jaeger, 2020-03-30:
> Can we say that this is NOT a bug in EKOS.
YES!

Chris-Rowland, 2020-03-30:
I don't know about the others but for the Celestron although the pier side that the mount reports is different to what INDI requires this is sorted out in the driver and the driver reports the pier side correctly.  Wolfgang says is is the same with the Avalon.  I don't know about this others but I don't think it matters.  As long as the driver can translate the pier side data that the mount provides to the correct pier side that INDI requires then everything should be fine.

As you say EKOS does not use the pier side to manage pier flip, it assues that a mount that slews to a position that is East of the meridian will be in the West (pointing East) pointing state and that at some point a little after it crosses the meridian a slew to the current position will do a pier flip.  If the mount has flipped early EKOS doesn't know. It will count down to the flip position and then do a slew.  This slew won't flip because the mount already has.  If the slew was too early EKOS doesn't know and no further flip slew is done, although one may happen as a result of a later alignment slew.

EKOS only handles the situation where slews to a position East of the meridian will leave the mount in the PierWest -> East state and to the West of the meridian in the PierEast -> West state.  Changing the Pier Flip state early or late is not handled.

Wolfgang,

I have been working on some code in the meridian flip part of EKOS which will read the PierSide from the mount and use that to manage the flip.  If a mount has already flipped it will do nothing.  If the flip fails then it will try again later. If pier side is not available it uses the current hour angle based method and hopes for the best.

I am also adding an Hour Angle Limit property that will stop the mount tracking if it tracks too far part the meridian in the wrong pointing state. 

I don't want to tread on your toes but could this be useful for getting pier flip to be more reliable for the mounts which correctly report the Pier side?

Chris

fdesvallees, 2020-03-30:
I did some more testing and I found that, as you both said, the pier side reporting has no effect on the decision to flip. 
My error was that I was not resetting the driver after changing the "Flip if HA>" value.
So the initial issue (pier side reporting) is closed. Sorry about that.

But for the rest ("Changing the Pier Flip state early or late is not handled") I agree with Chris. Let me know if I can help testing any code change.

François

sterne-jaeger, 2020-03-31:
@Chris-Rowland : don't worry, I'm happy to hear that I am not alone improving the meridian flip :-). Do you have your changes available in your repository? I'll send you a PM in the indilib forum so that we can discuss how to proceed.

fdesvallees, 2020-04-02:
I close this.

Chris, Wolfgang, for your info, I simply replaced the test `if (initialHA() >= 0` by
`if (currentTelescope->pierSide() == currentTelescope->PIER_EAST )` in ekos/mount/mount.cpp (and the reverse test a few lines below) and recompiled Ekos.

I tested limits of 0 and 10 degrees, and it works fine with TeenAstro.  I guess it will work for other mounts too. 

Note that you also need to edit indi/inditelescope.h to fix one bug:
replace
        typedef enum { PIER_UNKNOWN = -1, PIER_WEST = 0, **PIER_EAST = 0** } PierSide;
 by
        typedef enum { PIER_UNKNOWN = -1, PIER_WEST = 0, **PIER_EAST = 1** } PierSide;

Also, for testing the capture module must be up and running. Otherwise the flip is planned but never happens.

François

closed 2020-04-02 by fdesvallees