indilib/indi #1110: QHY5-II Color issue with the new releases

emanuelelaface, 2020-03-31:
I have a CCD QHY5-II Color and an AZ-EQ6 mount. I upgraded from 1.8.0  to 1.84 and the camera started to have many issues. The driver says No camera half of the times, when it starts the streaming is much slower, many times it hangs, if I stop and start the streaming most of the time it does not restart etc.

I tried to go back to 1.8.3 and I encountered the same issues, so I decided to rollback to 1.8.0 and everything works perfectly.

My system runs on a Pine64 that is a 64 bits arm system (aarch64) with the latest Armbian installed on it.

I run the server with the debug enable but I do not have any error, it simply has the issue that I reported but without triggering some specific error in the console.
Please let me know if I can do some additional test to debug this, I still have the 1.8.4 version available on my system, so I can do additional tentatives quickly.

knro, 2020-03-31:
This is most likely related to QHY SDK rather than driver. Can you git bisect and check if a particular libqhy version is the culprit?

emanuelelaface, 2020-03-31:
Thank you for the quick answer. I am a bit new to INDI, so I am not entirely sure about your suggestion.
To to install everything what I do is to get, compile and install indiserver, indi_eqmod, libqhy and indi_qhy_ccd. If I understand correctly you are suggesting to try various version of libqhy keeping the latest (1.8.4) for the other codes, am I right?
If so, I will try as soon as I can and report here.

knro, 2020-04-01:
Doing a git-bisect is a long involved processing requiring you to reinstall libqhy and then recompile indi-qhy after each step.

emanuelelaface, 2020-04-01:
ok, I can do it, this is not a big issue. My question was just to understand which component I should rebuild every time to test. I can do the libqhy and indi-qhy compile for several versions. I will let you know.

knro, 2020-04-01:
Ok hold for 30 mins while I build and upload the latest QHY SDK.

emanuelelaface, 2020-04-01:
Sure, I was having a coffe and I cannot compile anything before a coffe in the morning :-).

knro, 2020-04-01:
OK QHY SDK 20.3.21 is uploaded now.

emanuelelaface, 2020-04-01:
The SDK 20.3.21 is a bit better than the previous but still around 10 times slower than the one included in the version 1.8.0 (I think it is 5.1.0 but it seems that the numeration changed and now it uses the day of creation).
The version that started to generate this issue is the one marked (V2.1.2) on the QHYCCD site, (V2019.11.15.0) where there was a major USB change in order to fix some issue with USB 3 cameras. Unfortunately this creates some hiccups with my old QHY5-II color camera that worked perfectly before this change. The issue is a very slow frame rate (less than 1 per second), very slow in getting commands (it can take 30 seconds to stop a streaming and restart with a new exposure and more than 10 seconds to set a gain, sometime the gain is not set and the driver returns that the CCD_GAIN option is not available).
I will stay with the old driver and try the new SDK from time to time. Let me know if I can do more to fix this issue.

knro, 2020-05-12:
Please file this in INDI [3rd party](https://github.com/indilib/indi-3rdparty) repository.

closed 2020-05-12 by knro