indilib/indi #1125: EKOS/INDI/KStars crashes with build April 24 2020

jamiecflinn, 2020-04-25:
**Describe the bug**
Working system for past week using stable build- device:
LX200_ONSTEP
QHY 183M
ASI EFW
Pegasus PPB
Moonlight Focuser (arduino based MyFocuserPro2)

Installed latest update from stable build PPA dated 20200424
Kstars will run
Start EKOS - EKOS RUNS
Select Profile used in all operations -> EKOS crashes and so does Kstars - INDI server is still alive in background
Move to remote laptop and run UNUPDATED system and connect to REMOTE....all device connect! and can operate - only the LOCAL updated system cannot run
no error in logs with verbose turned on

**To Reproduce**
See above

**Expected behavior**
prior to update today full normal operation of EKOS/INDI

**Desktop (please complete the following information):**
 - OS: UBUNTU 18.04.4 LTS
 - indi-full is already the newest version (1.8.5~202004192150~ubuntu18.04.1).
kstars-bleeding is already the newest version (6:3.4.2+202004241139~ubuntu18.04
**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html
[log_22-08-41.txt](https://github.com/indilib/indi/files/4532198/log_22-08-41.txt)

TallFurryMan, 2020-04-25:
Well, in both cases you say INDI is working properly, so...this is not an INDI bug? Could you provide INDI logs instead of KStars ones? I think you are not posting on the right tracker.

knro, 2020-04-25:
Try to move the configuration directly elsewhere to test. i.e.

mv ~/.indi ~/backup_indi

then run, crash again? But as @TallFurryMan indicated, if this is a KStars crash (not INDI), then the right place for reporting is at bugs.kde.org

jamiecflinn, 2020-04-25:
Thanks Jasem...that solved the issue (moving config to back up) -

jamiecflinn, 2020-04-25:
there was another update today and whatever was not right is now OK after a number of full reboot tests with all devices - Thanks to who fixed something :-)

closed 2020-04-25 by jamiecflinn