indilib/indi #1128: indi_xagyl_wheel bug report

caggelis, 2020-04-27:
after compiling new drivers on Raspberry Pi ,running latest Astroberry distribution , I expect the following bug in xagyl_wheel driver.

While send to wheel any command , the driver going into a continues (non stop) loop trying to increase PulseWidth and the driver stucks and filter wheel stops working , giving "timeout error"
After that pulse get at the highest point (10000) and wheel cannot work again, until i send some "n" strings through terminal to the serial port , in case to reduce the pulse width to a normal point.

xagyl fw5125v3
raspberry pi 3 model b
astroberry 2.0.1

knro, 2020-04-27:
Thank you for the report, can you submit a PR to fix this issue?

caggelis, 2020-04-27:
fixed! 
replace following lines:
line 362:     (newPulseWidth - curPulseWidth)/100);
line 635:     int rc         = sscanf(res, "PulseWidth %duS", &pulseWidth);

commit 7894be27 (2020-04-28): Fix pulse width issue

Fixes #1128
closed 2020-04-28 by knro