indilib/indi #1138: SkySafari:  cannot slew to points less than 1 degree south of the equator (-0° issue)

setnes, 2020-05-03:
I'm using SkySafari Plus to control my telescope using the indi_skysafari driver.  Last night, however, I could not slew to objects within 1 degree south of the equator.

I tested using Zaniah (less than 1 degree South), and the mount ended up above the equator by the same amount as Zaniah is below, and SkySafari never appeared to complete the slew (had to click "Stop" afterwards).

I also tested with Porrima (more than 1 degree South), and the mount is able to GoTo the star without any issues.

The multiple tests that I did indicate that -0 degrees (with non-zero minutes/seconds) is converting to +0 degrees (with minutes/seconds).

I do not know if this bug is in the SkySafari INDI driver, or in SkySafari itself.  I also logged a similar issue with SkySafari.

Cartes du Ciel connected to INDI is able to slew to these points within 1 degree south of the equator without issue.

TallFurryMan, 2020-05-03:
This seems to be a standard DMS conversion issue. If I had time, I would check skysafari.cpp around line 531.

setnes, 2020-05-04:
Thanks for the reply.  After looking at that, I agree.  It looks like a bug.  Specifically, line 534...

`if (dd < 0)` 

Would replacing it with something like this fix it?

`if ( signbit(dd) )`

I'm making an assumption that what is passed in is a signed, negative zero and not an unsigned zero.  Also, I believe signbit would need `<cmath>` if it isn't already included somewhere.

closed 2020-05-04 by knro

setnes, 2020-09-30:
Assuming this change made it to Astroberry, I don't believe it was enough to fix the issue.  We are still unable to slew to areas between 0 and -1 degree using the SkySafari driver.  

Is there an additional spot in the code to look at that could require a similar negative zero fix?

commit 9f936d1b (2020-05-04): Fix -00 case as suggested by Kristopher.

Fixes #1138