indilib/indi #1139: Missing symbol in libindiclient.a

gnthibault, 2020-05-05:
Sorry, I think it should be pretty self explanatory:
nm -C /usr/lib/libindiclient.a | grep getText
000004ec T INDI::BaseDevice::getText(char const*)
         U INDI::BaseDevice::getText(char const*)
00000504 T INDI::Property::getText() const

Looks like this problem was not there in 1.8.4, but started to appea from 1.8.5, 1.8.6

gnthibault, 2020-05-05:
Sorry, just realized there is probably something I missed about dynamic method name resolution.

gnthibault, 2020-05-05:
Ok I tracked down the problem to this line:
https://github.com/indilib/indi/blame/62f2d19d489f45b7a0f84e664529e38421f5d4dd/libs/indibase/indiproperty.cpp#L272

and there:
https://github.com/indilib/indi/blame/62f2d19d489f45b7a0f84e664529e38421f5d4dd/libs/indibase/indiproperty.h#L76

A const was added
@knro  there is actually nothing wrong of course with main indi repository. but somehow, we need to make a change to pyindi-client, and I am not sure how to do that.

gnthibault, 2020-05-05:
I am trying to change
    result = (ITextVectorProperty *)(arg1)->getText();
to
    result = (const ITextVectorProperty *)(arg1)->getText();

in indiclientpython_wrap.cpp

gnthibault, 2020-05-05:
I cannot reproduce the problem on my other computer x86_64, although depending on the source (download tar from pypi or automatic pip build), I get either the const or non const version.
Maybe the problem is linked to the operating system/ABI ? that is not able to recognize both symbol as being one, this is outside of my expertise.

knro, 2020-05-05:
I just updated INDI::Property to be const-correct for all the constant functions. Can you purge any INDI libraries on your system and compile/install from latest GIT?

gnthibault, 2020-05-05:
Just rebuilt indi from
commit 1af2cbf3813854ffa2706e2ced43b72b023769cc
    Make rest of constant functions const

Nothing changed, 

Old libindiclient.a:
000004ec T INDI::BaseDevice::getText(char const*)
         U INDI::BaseDevice::getText(char const*)
00000504 T INDI::Property::getText() const

New libindiclient.a 
000004ec T INDI::BaseDevice::getText(char const*)
         U INDI::BaseDevice::getText(char const*)
00000504 T INDI::Property::getText() const

The swig shared object built by pyindi-client that I just rebuilt:
./build/lib.linux-aarch64-3.8/_PyIndi.cpython-38-arm-linux-gnueabihf.so | grep getText
000310e4 T INDI::BaseDevice::getText(char const*)
                  U INDI::Property::getText()
00036c64 T INDI::Property::getText() const

To me, it looks like the problem comes from the pyindi-client code. But Somehow I cannot reproduce the error on my x86_64 machine that also features different version of gcc, different OS, ..

gnthibault, 2020-05-05:
WAIT, I might have forgotten some sources in /usr/ instead of /usr/local

gnthibault, 2020-05-05:
That was it of course... Very sorry for any inconvenience. Thank you again for your support.

closed 2020-05-05 by gnthibault