indilib/indi #1143: GPhoto CCd with remote shutter

FredoTheGreat, 2020-05-12:
**Describe the bug**
When using remote shutter on ekos/kstars/gphoto ccd if the session involves a short exposure first (focus) then the camera goes into PC mode and the subsequent long exposure with the remote shutter gets ignored because it doesn't let go of the camera. it stays in pc mode

Camera is a Nikon D7000
USB device is a kmtronics device on /dev/ttyUSB0
Latest version of gphoto etc

To Reproduce
Start ekos first input /dev/ttyUSB0 then connect device and run a long exposure this will work fine for any exposure over 30 seconds
run short exposure (1s) the camera goes into PC Mode
then run long exposure (35s) relay on USB device fires but as camera is in pc mode and no shutter happens although an image is transfered to the folder its is garbage 
[log_20-34-14.txt](https://github.com/indilib/indi/files/4618057/log_20-34-14.txt)

When reproducing with the commandline tool, attach debugout using --debug --debug-logfile=debug.log if considered useful.
**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Perform action
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:
 - Version [e.g. 1.7.6, GIT..etc]

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

knro, 2020-05-12:
IIRC, this is fixed in nightly releases with FORCE_BULB set to ENABLED. Have you tried that?

FredoTheGreat, 2020-05-12:
FORCE_BULB is already enabled

FredoTheGreat, 2020-05-12:
in fact if force bulb was not enabled then the remote shutter would not work at all
the issue is that the remote shutter works until the camera takes an image less than 30 seconds in which it then uses the standard camera control and the camera goes into PC mode this would be ok if it released this when the job was complete so it could then use the remote shutter again

knro, 2020-05-12:
Yes, what you describe got fix in GIT already. Are you testing from stable repo?

FredoTheGreat, 2020-05-12:
I guess you mean a standard version and not any beta or nightly build so yes

FredoTheGreat, 2020-05-12:
is the nightly build available for raspbian?

FredoTheGreat, 2020-05-21:
Can you tell me the version of gphoto ccd driver that has this fix please I have version 2.9 at the moment which still has the issue

knro, 2020-05-21:
Are you running the GIT version?

FredoTheGreat, 2020-05-21:
I am running astroberry and stellarmate both show 2.9

Sorry I dont know how to run any other versions

knro, 2020-05-21:
For StellarMate, you use [Beta Channel](https://stellarmate.com/support.html?view=kb&kbartid=5) to get the latest nightly driver.

FredoTheGreat, 2020-05-25:
i tried the beta channel on stellarmate but couldn't get it to work as the gphotoccd crashes everytime i try to start it

knro, 2020-05-25:
I just tested on the beta channel and it works fine with both Canon 600D and Nikon 3500. Can you run it via gdb? Open a console and type in this:

igdb indi_gphoto_ccd indi_simulator_telescope

then in Ekos create a profile, set host to **remote** and select GPhoto and Telescope Simulator. Make sure INDI Web Manager is not checked and then try to connect. If it crashes, go back to the console. If it crashes, you should see a prompt, type in **bt** to get the backtrace and post it here.

FredoTheGreat, 2020-05-25:
When I run that line i get this response:

stellarmate@stellarmate:~ $ igdb indi_gphoto_ccd indi_simulator_telescope
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from indiserver...Reading symbols from /usr/lib/debug/.build-id/04/f987ebace88febf7431b662ff246b6664a07b1.debug...done.
done.
Starting program: /usr/bin/indiserver -v indi_gphoto_ccd indi_simulator_telescope
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
2020-05-25T15:31:03: startup: /usr/bin/indiserver -v indi_gphoto_ccd indi_simulator_telescope 
[Attaching after Thread 0xb6ffa090 (LWP 3257) fork to child process 3260]
[New inferior 2 (process 3260)]
[Detaching after fork from parent process 3257]
[Inferior 1 (process 3257) detached]
2020-05-25T15:31:03: Driver indi_gphoto_ccd: pid=3260 rfd=3 wfd=6 efd=7
2020-05-25T15:31:03: Driver indi_simulator_telescope: pid=3261 rfd=4 wfd=9 efd=10
2020-05-25T15:31:03: listening to port 7624 on fd 5
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
process 3260 is executing new program: /usr/bin/indi_gphoto_ccd
2020-05-25T15:31:03: Driver indi_simulator_telescope: HaAxis: TrackRate 1, trackingRateDegSec 15.041067 arcsec
2020-05-25T15:31:03: Driver indi_simulator_telescope: snooping on GPS Simulator.GEOGRAPHIC_COORD
2020-05-25T15:31:03: Driver indi_simulator_telescope: snooping on GPS Simulator.TIME_UTC
2020-05-25T15:31:03: Driver indi_simulator_telescope: snooping on Dome Simulator.DOME_PARK
2020-05-25T15:31:03: Driver indi_simulator_telescope: snooping on Dome Simulator.DOME_SHUTTER
2020-05-25T15:31:03: Driver indi_gphoto_ccd: indi_gphoto_ccd: symbol lookup error: indi_gphoto_ccd: undefined symbol: _ZN4INDI7CCDChip8setFrameEiiii
2020-05-25T15:31:03: Driver indi_gphoto_ccd: stderr EOF
2020-05-25T15:31:03: Driver indi_gphoto_ccd: restart #1
2020-05-25T15:31:03: Driver indi_gphoto_ccd: pid=3262 rfd=0 wfd=7 efd=8
Child process 3260 died
[Inferior 2 (process 3260) exited with code 0177]
(gdb) 2020-05-25T15:31:03: Driver indi_gphoto_ccd: indi_gphoto_ccd: symbol lookup error: indi_gphoto_ccd: undefined symbol: _ZN4INDI7CCDChip8setFrameEiiii
2020-05-25T15:31:03: Driver indi_gphoto_ccd: stderr EOF
Child process 3262 died
2020-05-25T15:31:03: Driver indi_gphoto_ccd: restart #2
2020-05-25T15:31:03: Driver indi_gphoto_ccd: pid=3263 rfd=0 wfd=7 efd=8
2020-05-25T15:31:04: Driver indi_gphoto_ccd: indi_gphoto_ccd: symbol lookup error: indi_gphoto_ccd: undefined symbol: _ZN4INDI7CCDChip8setFrameEiiii
2020-05-25T15:31:04: Driver indi_gphoto_ccd: stderr EOF
Child process 3263 died
2020-05-25T15:31:04: Driver indi_gphoto_ccd: restart #3
2020-05-25T15:31:04: Driver indi_gphoto_ccd: pid=3264 rfd=0 wfd=7 efd=8
2020-05-25T15:31:04: Driver indi_gphoto_ccd: indi_gphoto_ccd: symbol lookup error: indi_gphoto_ccd: undefined symbol: _ZN4INDI7CCDChip8setFrameEiiii
2020-05-25T15:31:04: Driver indi_gphoto_ccd: stderr EOF
2020-05-25T15:31:04: Driver indi_gphoto_ccd: restart #4
Child process 3264 died
2020-05-25T15:31:04: Driver indi_gphoto_ccd: pid=3265 rfd=0 wfd=7 efd=8
2020-05-25T15:31:04: Driver indi_gphoto_ccd: indi_gphoto_ccd: symbol lookup error: indi_gphoto_ccd: undefined symbol: _ZN4INDI7CCDChip8setFrameEiiii
2020-05-25T15:31:04: Driver indi_gphoto_ccd: stderr EOF
2020-05-25T15:31:04: Driver indi_gphoto_ccd: restart #5
2020-05-25T15:31:04: Driver indi_gphoto_ccd: pid=3266 rfd=0 wfd=7 efd=8
Child process 3265 died
2020-05-25T15:31:04: Driver indi_gphoto_ccd: indi_gphoto_ccd: symbol lookup error: indi_gphoto_ccd: undefined symbol: _ZN4INDI7CCDChip8setFrameEiiii
2020-05-25T15:31:04: Driver indi_gphoto_ccd: stderr EOF
2020-05-25T15:31:04: Driver indi_gphoto_ccd: restart #6
Child process 3266 died
2020-05-25T15:31:04: Driver indi_gphoto_ccd: pid=3267 rfd=0 wfd=7 efd=8
2020-05-25T15:31:04: Driver indi_gphoto_ccd: indi_gphoto_ccd: symbol lookup error: indi_gphoto_ccd: undefined symbol: _ZN4INDI7CCDChip8setFrameEiiii
2020-05-25T15:31:04: Driver indi_gphoto_ccd: stderr EOF
2020-05-25T15:31:04: Driver indi_gphoto_ccd: restart #7
2020-05-25T15:31:04: Driver indi_gphoto_ccd: pid=3268 rfd=0 wfd=7 efd=8
Child process 3267 died
2020-05-25T15:31:04: Driver indi_gphoto_ccd: indi_gphoto_ccd: symbol lookup error: indi_gphoto_ccd: undefined symbol: _ZN4INDI7CCDChip8setFrameEiiii
2020-05-25T15:31:04: Driver indi_gphoto_ccd: stderr EOF
2020-05-25T15:31:04: Driver indi_gphoto_ccd: restart #8
Child process 3268 died
2020-05-25T15:31:04: Driver indi_gphoto_ccd: pid=3269 rfd=0 wfd=7 efd=8
2020-05-25T15:31:04: Driver indi_gphoto_ccd: indi_gphoto_ccd: symbol lookup error: indi_gphoto_ccd: undefined symbol: _ZN4INDI7CCDChip8setFrameEiiii
2020-05-25T15:31:04: Driver indi_gphoto_ccd: stderr EOF
[... cut ...]
2020-05-25T15:31:04: Driver indi_gphoto_ccd: stderr EOF
2020-05-25T15:31:04: Driver indi_gphoto_ccd: restart #10
Child process 3270 died
2020-05-25T15:31:04: Driver indi_gphoto_ccd: pid=3271 rfd=0 wfd=7 efd=8
2020-05-25T15:31:04: Driver indi_gphoto_ccd: indi_gphoto_ccd: symbol lookup error: indi_gphoto_ccd: undefined symbol: _ZN4INDI7CCDChip8setFrameEiiii
2020-05-25T15:31:04: Driver indi_gphoto_ccd: stderr EOF
2020-05-25T15:31:04: Driver indi_gphoto_ccd: Terminated after #10 restarts.
Child process 3271 died
(gdb)

knro, 2020-05-25:
sudo apt-get remove indi-gphoto
sudo apt-get update && sudo apt-get -y install indi-gphoto

FredoTheGreat, 2020-05-25:
ok i have edited the gphoto CCD_config.xml to add the /dev/ttyUSB0 which has stopped it crashing 
the system does indeed enable the USB switch and fires the shutter.
however the images are not transferring correctly and the exif info shows shutter duration as 1sec when it was supposed to be 35
the images are very poor and indeed look like they are only 1sec so i am unsure as to where the data is being lost as the shutter does open for 35 seconds.
if i use my manual invalometer it works well and the images (saved on the camera sd card) are good.

gubihu, 2020-05-31:
I'm also using a Nikon D7000 and I have never succeeded to take an expo longer than 20 seconds.
When I try to take (say) 30 sec expo here what happens:
I hear the shooter open and about 30 sec later close, but I got no picture:
2020-05-31T16:06:07: [ERROR] Exposure failed to save image... File not found 
2020-05-31T16:06:07: [ERROR] Error downloading image from camera: File not found 
2020-05-31T16:05:27: [INFO] Starting 30 seconds exposure. 
I have just tried it with the latest checkout from github just compiled.

For 20 sec expo it sometimes succeeds, sometimes fails as above.
For 16s sec it mostly works, and bellow 10 sec it always (practically) works.

For longer than 30 the machine actually takes 30 sec expo, the driver waits for the given time to elapse, the fails with the above message.
It seems to me, that the pictures are not taken by bulb time, and the "Force BULB" option does not seems to be working.

FredoTheGreat, 2020-05-31:
Have you got it set to save to sd

gubihu, 2020-05-31:
I use the default settings for Capture Target,which is RAM, IT does work for exposure shorter than about 20 sec. No SD card is needed in the Nikon.

Now, as you have asked, I've tried to set Capture Targer to SD Card. It worked, I'm able to take 30 sec exposures. 
But longer than 30 sec still does not work:
DEBUG   693.465817 sec  : Finding optimal exposure setting for 40 seconds in shutterspeed2 (count=57)...
DEBUG   693.465905 sec  : Closest match: 30 seconds Index: 54
DEBUG   693.465981 sec  : Setting radio/menu widget shutterspeed2: 54 (30)
DEBUG   693.469824 sec  : Setting new configuration OK.
DEBUG   693.469964 sec  : Using predefined exposure time: 30 seconds
DEBUG   693.470051 sec  : Exposure started.
DEBUG   733.370704 sec  : Reading exposure...
DEBUG   733.370784 sec  : Exposure complete.
DEBUG   733.370812 sec  : Downloading image... Name: (BUD_2239.NEF) Folder: (/store_00010001/DCIM/103D7000) Delete from SD card? (false)
DEBUG   734.784034 sec  :  Downloaded 0x0 (preview 0x0)
DEBUG   734.851263 sec  : Capture target is SD Card.
INFO    734.851481 sec  : Exposure done, downloading image...
DEBUG   736.474986 sec  : read_libraw: raw_width: 4992 top_margin 0 left_margin 0 first_visible_pixel 0
DEBUG   736.475135 sec  : read_libraw: rawdata.sizes.width: 4948 rawdata.sizes.height 3280 memsize 32458880 bayer_pattern RGGB
DEBUG   736.513663 sec  : read_libraw: memsize (32458880) naxis (2) w (4948) h (3280) bpp (16) bayer pattern (RGGB)
DEBUG   736.721505 sec  : Uploading file. Ext: fits, Size: 32466240, sendImage? Yes, saveImage? No
DEBUG   737.236036 sec  : BLOB transfer took 0.513821 seconds
DEBUG   737.236139 sec  : Upload complete

Here I was asking for a 40 sec exposure, and what I got is a 30 sec exposure (which is the longest a D7000 can do without using BULB).

The problem with the RAM pictures is clearly related to some timing problem. Since it is taken using preset timing on the Nikon, the exposure is measured by the Nikon, and also by the driver, but the two timing does not agree exactly. The driver tries to access the picture either too early or too late (I'm not sure). It seems to be more serious the longer the exposure.

FredoTheGreat, 2020-05-31:
It must be set to sd card

The d7000 is limited to 30sec max via the usb port that’s the way it is

You need to use an external shutter release system if you want more

gubihu, 2020-06-01:
I have seen a thread about a similar problem on the gphoto github:
https://github.com/gphoto/libgphoto2/issues/495

knro, 2020-06-06:
So I take this setting to **SD Card** does not have the timing issue with latest libgphoto2?

FredoTheGreat, 2020-06-06:
not related imo
you have always needed to set the SD card otherwise it will not work at all 
the issue I have is for exposures over 30 secs using the remote shutter

gubihu, 2020-06-06:
That is not the case!
It works with internal RAM unless the expo is long. I have taken hundreds of images using 4s, 8s and 16s expos. All to RAM. No SD card. It worked without problem.
20 sec expo sometimes works, sometimes not. 30 sec => always error.

I have also got a definit answer about bulb time on ghoto2 github:
D7000 lacks an important API call, "TerminateCapture" , that is why bulb timing and longer than 30 secs does not work. D7100 has it, D7000 lacks. Too bad. But it has nothing to do with the SD card.

But you are right: "The d7000 is limited to 30sec max via the usb port that’s the way it is".
Can you tell me what kind of remote shutter cable do you use. Where did you get it from? I cannot find any on the web (just DSUSB, but that is something different, not ttyUSB).

I'm thinking about connecting an optocoupler to one of the Raspberry PI GPIO port, and triggering the shutter from there via a standard 3.5mm jack to shutter cable. It could be easy enough  to modify the driver to support that.

FredoTheGreat, 2020-06-06:
not if you use the external shutter control but it is a moot point the original issue I listed is for exp over 30 seconds using the remote shutter control.
the one I have made is with the KMtronic usb relay here https://info.kmtronic.com/usb-one-channel-relay-box.html

gubihu, 2020-06-06:
I see, thanks.

FredoTheGreat, 2020-06-11:
this appears to now be fixed. well it is on astroberry so I am guessing it will on Stellarmate as well

closed 2020-06-11 by FredoTheGreat