indilib/indi #1146: Duplicated code in base drivers that use indicom

pardo-bsso, 2020-05-15:
Hi all,

after much head scratching the other week while doing work on the talon6 driver I noticed that the same code around indicom to handle connections is repeated, see for example

https://github.com/indilib/indi/blob/master/libs/indibase/inditelescope.cpp#L1559
 and
https://github.com/indilib/indi/blob/master/libs/indibase/indidome.cpp#L2245

or

https://github.com/indilib/indi/blob/master/libs/indibase/indidome.cpp#L225
and
https://github.com/indilib/indi/blob/master/libs/indibase/inditelescope.cpp#L232

This is also true for indifilterwheel, indifocuser, indirotator, indisensorinterface and indiweather.
Is there any reason for this? I'll have some time in the next days to refactor this behaviour into a single class if nobody objects.

Thanks

knro, 2020-05-16:
Not sure, maybe it was there to separate concrete drivers (e.g. INDI::Focuser) from interface classes (INDI::FocuserInterface) ? You can have a Telescope driver with interfaces of other device subtypes, and due to multiple inheritance it can get messy. I'd be careful about changing this unless there is a noticable net gain.

pardo-bsso, 2020-05-16:
Perhaps I'm not seeing something of the big picture here.

I was more thinking about getting all the common stuff like setting the connection type, handshakes, etc inside DefaultDevice. Right now the concrete implementations already call set[device-type]Connection() when they need or require only a specific type and not all of them.

But on the other hand everything works as it is now, so if nobody else chimes in let's cut this discussion for later.

Thanks

closed 2020-08-30 by knro