indilib/indi #1155: indi_eval memory leak

setnes, 2020-05-29:
When using indi_eval to wait for an indi parameter to change, more and more memory is consumed.

To reproduce, issue an indi_eval command at the prompt that waits for a value to change... like this.

`indi_eval -t 0 -w  '"GPSD.TIME_UTC._STATE"==3'`

Then find the process ID associated with the indi_eval statement.  

`ps -ef|grep indi_eval`

Use the `pmap` command against the process ID to monitor the memory usage.  The total memory used will slowly go up, and never down.

The expected behavior is that you can use indi_eval to trigger something to happen at the OS level based on an indi parameter reaching a certain value.  For events that do not occur frequently, this memory leak can consume enough memory over time to affect other processes.

OS: Raspbian Buster (Astroberry)
indi_eval version: 1.5

knro, 2020-05-30:
@setnes Thank you for the report, any pointers to the memory leak?

setnes, 2020-05-30:
I assume it's something in tools/evalINDI.c, but I really don't know what to look at.

TallFurryMan, 2020-05-30:
I don't think it is a leak. The process will read the XML stream until it finds the desired element, but doesn't seem to release the data already processed. It could indeed be an improvement to cleanup when the argument timeout is infinite.

setnes, 2020-05-30:
So how is that not a leak?  :)  

It might be just a matter of semantics, but the bug makes indi_eval not usable on systems with limited memory when monitoring events that do not frequently happen.

TallFurryMan, 2020-05-31:
It's not a leak because the memory block is still held by the program. Agreed, that's semantics :)

A simple improvement would probably be to release and allocate a new XML analyzer each time a full message is received.

knro, 2020-08-30:
Any update on this issue?

rseidler2019, 2021-05-07:
I am not seeing any strange memory behaviour of indi_eval on my Ubuntu laptop. Is the problem still occurring on other OS or is it fixed in a newer release? If it is still a problem, please describe your setup in more details (started indiserver/indiweb, used drivers, etc.). Thanks

closed 2024-04-23 by github-actions[bot]