indilib/indi #1158: indi_skywatcherAltAzMount does not connect wirelessly

federicoparra, 2020-06-02:
**Describe the bug**
Using exactly the same parameters than I use for connection in indi_azgti_telescope driver: 
192.168.43.195
11880
UDP
The driver attempts to connect but nothing happens.
The mount and the Raspberry Pi 4 (w/Astroberry server) connects to my phone's hotspot (192.168.43.1).
If I do the same with indi_azgti_telescope it connects immediately.
Last version of Astroberry (Version 2.0.1)

knro, 2020-06-06:
@kecsap is this driver capable of connection via UDP? I don't recall that it supports that explicitly.

kecsap, 2020-06-06:
Not really, I don't know what are the requirements. It was only tested with serial connection.

knro, 2020-06-06:
@federicoparra is there any reason you're using this over AZ Gti?

federicoparra, 2020-06-06:
Hi, the option to connect through UDP is there. I'm using it because I want to use AZ-GTI connecting directly to the mount (i.e., without using synscan as intermediary) and in Alt-az mode. The Az-Gti indi driver connects through UDP directly but doesn't allow to use Alt-az mode :( the synscan indi driver allows Alt-az...but only using the synscan app as intermediary. Also, I read here that this driver could be made to with through UDP with Az-gti for Alt-az control: https://indilib.org/forum/general/6419-sky-watcher-az-gti-alt-az-mode-can-be-used.html

kecsap, 2020-06-06:
@federicoparra As far as I remember from the past, the Synscan driver put to PC Direct Mode is equivalent to a simple USB cable with serial adapter. You don't need UDP for these modes, UDP only needed for wireless (WiFi) connection.

federicoparra, 2020-06-06:
I'm always talking about wifi. I don't use cables.

federicoparra, 2020-06-06:
UDP is required for az-gti to connect wirelessly. It works flawlessly with Az-gti indi driver which connects directly to the mount in eq mode (it's a mod of eq-mod indi driver) and also works well with synscan driver, alas you need to have the app working somewhere e.g. on your phone and connect to the app wirelessly through UDP (a different port than for direct connection to the mount)

kecsap, 2020-06-06:
Then I can't help. I don't have this kinda mount, you can only try what T-Studio did in his topic or the best comment on his topic and ask for help to achieve the same.

federicoparra, 2020-06-06:
For some weird reason I can't login to indilib.org to comment on forums. Tried creating two accounts but always get login error after account creation. Tried different browsers. Made a post about this in Cloudynights with no response. In fact that's why I ended posting here. Anyways, I don't understand what he suggests on that post either :( is he talking about changing the driver's code?

knro, 2020-06-13:
@federicoparra can you please update from latest Github and test with wireless connection again?

gubihu, 2020-06-13:
I have already reported this issue here:
https://indilib.org/forum/mounts/6467-indi-skywatcheraltazmount-wireless-handshake-failed.html
and we were discussing (among other things) here:
https://indilib.org/forum/ekos/6468-indi-skywatcheraltazmount-shown-in-wrong-position.html

Since that I have investigated the code more in depth, and I've found that the problem is how 
SkywatcherAPI::TalkWithAxis() 
handles the connection: 
    while (!EndReading)
    {
        std::this_thread::sleep_for(std::chrono::milliseconds(5));
        char c;

        int rc = skywatcher_tty_read(MyPortFD, &c, 1, 10, &bytesRead);
....
    }
That is calls skywatcher_tty_read() with nbytes=1 in a cycle, which calls indicom.c : tty_read() also with nbytes=1. That works for serial connection, but for UDP it seems it reads the first byte of the UDP packet and discards the rest.
So one of these functions in the call chain needs a major rewrite for that to work. 
:(

knro, 2020-06-13:
@gubihu So in this commit 2e5327ea211044fca6356ae3bf06df8be283620d I added UDP support to the driver. That does not help?

gubihu, 2020-06-13:
@knro I see now, it is in skywatcherAltAzSimple but not in skywatcherAltAzMount. 
But unfortunatelly it does not help. I've just tried.
In  indicom.c : tty_read() there are some if (tty_gemini_udp_format) parts, 
but no if (tty_generic_udp_format).

It fails in tty_read() line 449 on
...
        if ((err = tty_timeout(fd, timeout)))
            return err;
...
 when tty_read() called for the second time to read the second byte,

knro, 2020-06-13:
@gubihu Any chance tty_generic_udp_format can be added there as well?

gubihu, 2020-06-13:
@krno Do you mean adding tty_generic_udp_format to tty_read()?
No,  I don't think so. I mean it would not help.
What is in tty_read() for tty_gemini_udp_format seems to be that in case of gemini (I have no idea what is gemini) there must be an 8 byte sequnce number at the head of every answer packet, and it just check and cut that 8 byte.

It will not help on the fact that SkywatcherAPI::TalkWithAxis() tries to read byte by byte. I think (but not sure) that it is SkywatcherAPI::TalkWithAxis() that needs to be fixed.

gubihu, 2020-06-13:
It should utilize tty_read_section() just as it is done in Skywatcher::read_eqmod().

knro, 2020-06-13:
@gubihu Alright, I guess you went ahead and testing this change now... looking good so far?

gubihu, 2020-06-13:
@knro Actually yes, it is able to connect and move the mount.
There are still things to do, this part of the code is complicated.

federicoparra, 2020-06-13:
Amazing that now it's working! Newby question, how do I install the latest version of the driver? I'm using astroberry 2.0.2 (latest build)

gubihu, 2020-06-13:
It is not in the repo yet. It is in my own repo (https://github.com/gubihu/indi.git), but at the moment UDP works, serial does not.

gubihu, 2020-06-13:
@knro It works now both with serial and UDP connection. It is in my fork (https://github.com/gubihu/indi.git), under sw_api_udp branch. 
It is not a trivial change, I had to add an other virtual method to skywatcherAPI to be consistent with the existing code, but it may brake other possible drivers using skywatcherAPI.
Do you want to take a look at it before I send a pull request?

knro, 2020-06-13:
@gubihu Please submit the PR and we'll review it there.

closed 2020-06-14 by knro

gubihu, 2020-06-14:
@federicoparra Now, that my patch has been merge, you can check out and compile the indi software for yourself. Detailed instructions on how to compile are in the main README file of the repo.

federicoparra, 2020-06-19:
Sorry for a newby question: I use Astroberry (2.0.2), can I expect an automatic update or I do need to compile this myself? Is this now part of the "normal" branch or a separate branch? thanks!

gubihu, 2020-06-20:
It is in the master branch. Sooner or later it will find it's way to the released packages.

federicoparra, 2020-07-12:
> It is in the master branch. Sooner or later it will find it's way to the released packages.

I don't understand, I run sudo apt install indi-full gsc and also sudo apt update && sudo apt upgrade && sudo apt dist-upgrade and let that run, all got updated, but the driver still does not connect to AZ-GTI. I can perfectly connect with the AZ-GTI EQ driver (based on EQ mod). Version of the indi_skywatcherAltAzMount is 1.0, should it be something else? What can I do to know if the version I have has the updates you made? also running the sudo commands I listed above I should have gotten the update right?

gubihu, 2020-07-12:
What you get by apt is the so called 'released binary package'. I have no idea about the release schedule. In the mean time you have to compile from source to get the bleeding edge version. The instructions for compile is here in the top level readme of this project: https://github.com/indilib/indi/blob/master/README.md

federicoparra, 2020-07-13:
> What you get by apt is the so called 'released binary package'. I have no idea about the release schedule. In the mean time you have to compile from source to get the bleeding edge version. The [...]

Do those instructions apply to drivers? I'm a Newby but reading it I get the sense is just for the core functionality? Also Raspbian (Raspberry OS) is not in the list, would it work? Isn't there an easier way to update "just" this driver with your modified version?

federicoparra, 2020-07-15:
> Do those instructions apply to drivers? I'm a Newby but reading it I get the sense is just for the core functionality? Also Raspbian (Raspberry OS) is not in the list, would it work? Isn't there an [...]

Awesome I did and it works perfectly now!!

gubihu, 2020-07-15:
Sorry for missing to answer. Yes, it compiles just fine undes Raspbian. And yes, this driver is actually part of the core library.

PR #1172 (2020-06-14): UDP support for skywatcherAPI
SkywatcherAPI::TalkWithAxis() changed to use tty_read_section() to read the response for a command. This way both Serial and UDP connection can work. Fix #1158