indilib/indi #1159: Plate solving does not automatically iterate using indi_synscan_telescope in alt-az with Az-GTI mount

federicoparra, 2020-06-02:
Equipment:
-Nikon D5100
-Raspberry Pi 4 / 2GB
-Astroberry 2.0.1
-AZ-GTI / Synscan Pro app, 1.19

Connecting Ekos to Synscan via indi_synscan_telescope in alt-az
Works perfect for goto and tracking and plate solving BUT does not iterate, i.e., it solves, slew then stops. I can plate solve again manually several times no problem but does not iterate automatically.

knro, 2020-06-14:
Please attach the complete log (including Alignment + INDI + Telescope)

federicoparra, 2020-06-19:
Could you let me know how to get those logs? I will produce them for you. Thank you! 
BTW, I think (because of your picture) that I recognize you from the Indilib/Ekos/Stellarmate team? If yes, please, can you help me, I have tried creating 3 different accounts to post on the Indilib forums and with all three of them I get "login failed" and can get in to comment on the forum. That's why I ended commenting here...

knro, 2020-06-19:
I checked your account and it's already active.

federicoparra, 2020-06-19:
[image]
Please check this screenshot, "login failed"

federicoparra, 2020-07-01:
> I checked your account and it's already active.

Dear Knro, have you seen my screenshot above? I can't log in :(

knro, 2020-09-28:
Is this still an issue?

federicoparra, 2020-09-28:
I'll check this weekend.

federicoparra, 2020-10-07:
@knro  tried today with the CCD simulator but it doesn't work because alignment is always perfect and no need for more plate solving cycles. Any way I can try this with the CCD simulator ? otherwise I'll have to keep waiting for open skies

federicoparra, 2020-11-10:
@knro tested tonight works great now! Good enough solution to use this driver with synscan on same Android phone for good enough alt-az operation

federicoparra, 2020-11-18:
I spoke too quickly. As I mentioned in private on an email today, @knro for some reason the driver asks for confirmation every time when aligning, before slew, mentioning the desired slew destination is below the horizon, which it isn't. If one accepts, it does a small slew as expected, not sure that in the end the object would get centered though. 
[image]

federicoparra, 2020-11-30:
Hi! any news about this?

knro, 2020-12-09:
I made recent changes to SkyWatcher Alt-Az driver and it should work with AZ-GTi driver as well in Alt-Az mode. Please test that. In StellarMate, you can change to beta channel or simply open the terminal and type in this command:

`update_indi_core`

federicoparra, 2020-12-09:
Hi @knro, I'm confused as we are talking about three different things: this bug report is specifically about Synscan driver, and it's issue with plate solving when connected to AZ-GTI mount. The problem used to be that it would not iterate - that is solved, it does iterate now, but after each iteration, a new bug arises, which is that (as per the screenshot above) it messes up with the location, thinking that the telescope will end pointing below the horizon when it's clearly not, and even if we accept, the next movement does not really improve (the solver does not seem to go to a useful position from the previous one). 
Then, separately, there is the issue of the AZ-GTI driver, which is awesome and does not need the Synscan app to work, but that so far it was not working in Alt-Az mode and we were waiting for the driver's author to complete an Alt-Az compatible version. Finally, there is the Skywatcher Alt-Az driver, an old driver that I tried and that worked but that had a very basic implementation that was not good enough for photography or in some cases even for viewing (keeps slewing all the time at intervals, instead of simply tracking). 

So, what are you referring to in your message above? Have you corrected the Synscan driver platesolving problem? Or has the AZ-GTI driver been completed / added an Alt-Az mode? Or have you corrected the Skywatcher Alt-Az driver so that it does not anymore slew all the time and instead track correctly? 

Please do let me know as I will be using the mount extensively during the next month and it will be thus the perfect time for all sorts of tests!

knro, 2020-12-09:
I fixed the issues in the SkyWatcher Alt-Az driver. No changes were made to Synscan driver, it's kept as-is, it's a minimal driver and is not expected to work as well as the low-level SkyWatcher driver.

So in essence, forget all the other drivers and just use the SkyWatcher Alt-Az driver now. Please report back on how it goes, thanks!

federicoparra, 2020-12-09:
Oki! I'll try that driver again soon, flying to my homeland (Argentina!) for a month, I'll have clear skies there (it's summer there!)

federicoparra, 2021-02-24:
SkyWatcher Alt-Az driver (also called Az-Gti Alt-Az) still doesn't work as intended:
1) when directing it up, down, left or right using stellarmate manual slew controls, once the button is released, the mount start slewing randomly in some random direction/s for about 10 seconds
2) when tracking, tracking is not accurate compared to the Synscan driver - Synscan driver allows for 10 second exposures without blurring whereas this Alt-Az driver doesn't allow even 3 second exposures without blurring.

closed 2021-04-17 by knro

knro, 2021-05-28:
This should be fixed in master.