indilib/indi #1160: Skywatcher Synscan wrong Longitude

Mike211-Sky, 2020-06-02:
I have KStars running inside StellarMate on a Raspberry PI 3 and a Skywatcher HEQ5 Mount running SynScan.

When you set a geographic Location in KStars or use a GPS module with a West Longitude (negative) and have Indi updates turned on, my Synscan is set incorrectly to an East Longitude of the same value. KStars is reporting the correct Geographic Location confirmed by star locations.

Exact steps to reproduce the behavior:
1. Login to remote Stellarmate using wifi from laptop
2. Open KStars>Settings>Configure KStars... then under INDI tab set the following:
  'KStars Updates All Devices' Selected
  'Time' Ticked
  'Location' Ticked
2. Change Geographic Location to something with a West Longitude (Southampton, United Kingdom)
3. Select Start Profile on Ekos Setup screen
4. Check Location in SynScan hand controller - Setup Menu>Observ Site
5. Select Disconnect on Ekos Setup screen
6. Select another Geographic Location
7. Select Connect Profile on Ekos Setup screen
8. Check Observing Site in SynScan hand controller.

All locations selected always give an East Longitude in SynScan, even though the values are shown correctly as negative in KStars.

Equipment versions used:
 - Windows 7 Laptop using VNC Viewer in Google Chrome
 - Raspberry PI 3 running Stellarmate OS 1.5.2
 - KStars 3.4.2
 - SynScan 03.39.10

TallFurryMan, 2020-06-02:
Could it be that you are under one degree of the Greenwich meridian? This must be yet another less-than-one-degree DMS conversion issue...

knro, 2020-06-06:
@Mike211-Sky @TallFurryMan makes a good point.. can you share your longitude?

Mike211-Sky, 2020-06-07:
My Longitude is 2 degrees West.

The bug I detailed above relates to any West Longitude entered and has nothing to do with the less-than-one degree issue. I tested it with multiple selections from locations around the world listed within KStars.

knro, 2020-06-07:
and in INDI Control Panel it's written as longitude 358?

Mike211-Sky, 2020-06-08:
I am no code expert, but on investigating the bug in synscandriver.cpp, I think I have found the error.

line 1170 reads:  cmd[9] = IsWest ? 1 : 0;

Should this not be cmd[8], as the next byte to send?

commit 8ad96012 (2020-06-09): Fixes #1160

Thanks to @Mike211-Sky for pointing our the error in sending the location command.
closed 2020-06-09 by knro