indilib/indi #1162: EQMod EQ8-R "autohome" bug/driver update required

xthestreams, 2020-06-06:
**Describe the bug**
The Autohome feature does not work with the new EQ8-R as there are no Aux encoders per their product information  "Note: Freedom-Find encoders are not present on the EQ8-R" 

However the mount is now capable of Autohome with a different set of sensors - I have confirmed this through the handset which allows me to "Autohome" and ALSO to set a parking "offset" DEC eg: 90" (as different from "home").

The feature does not work with the INDI EQMod Autohome button in the INDI Control Panel, I assume because the driver is looking for the encoders from the old EQ8 which makes sense, but as they are not present on the EQ8-r (and H?) - it seems a new approach is required.

Anything you could do to enable the Autohome feature to work on the EQ8-r would be awesome. I guess I am the first, I can't see anyone else having this problem on the forums.

**To Reproduce**
1. Purchase an EQ8-R (or get a friend to buy one, borrow it and never give it back)
2. plug into RPi, setup as a new mount - set baud rate to 115200, etc
3. align, sync, tracking enabled.
4. park/click Autohome and an error appears after the "warning" message indicating that the encoders are not present - which is true but not required in the EQ8-R 

**Expected behavior**
- That it should run the Autohome routine and get back to a sane place without me needing to go to the observatory.
- Ideally should also implement the "offer" feature to Autohome and then move to a "post-park" position whilst sleeping (eg: Dec -90")
- Upon Unpark should Autohome prior to doing PA, etc

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Mac OS plus Stellarmate RPi 4
- current

**Log Files**
N/A

knro, 2020-06-06:
@geehalel Hello Jean-Luc, any ideas how to handle this case in the driver?

geehalel, 2020-06-06:
Hello,
I remember that I implemented the autohome feature based on a document sent by someone here (or in the old forum), who received it from a skywatcher technician (maybe for the ASCOM eqmod driver). I had no idea of how to do it, and if you look at the driver code it was quite complex.
Thus at first sight it may be very hard to do without any information from skywatcher or from someone who reverse engineered this new autohome algorithm. 
I just looked on the skywatcher site and only found the motor controller firmware. It seems this is an ARM cortex binary file,  quite hard to emulate mainly because of peripherals.
Maybe someone can look at the messages between the hand controller and the mount while autohoming, at least this could give us an idea of the algorithm.

xthestreams, 2020-06-06:
I'll go to Skywatcher now and see what I can find out. Had a quick look at EQMod for ASCOM, nothing I can see there.

AstroOlly, 2020-06-12:
I don’t understand this issue, the EQ8-r has the same autohome sensors fitted as the older EQ8, these are completely independent of the Aux encoders, (Which the RQ8-r does not have) so why does it not work because there are no encoders....it does not make sense....

xthestreams, 2020-06-12:
Thanks Astronerd - let's hope it's a simple fix!

AstroOlly, 2020-06-12:
@geehalel 
I am not a coder at all, but the issue is that driver looks for the encoders before it performs the autohome function, this was fine on the old mounts that had them, but the new EQ8-r does not, so can it be changed not to look for the encoders before it performs the autohome function...as the AH has its own sensors located on the DEC and RA shaft, and so it does not use the encoders to do this...
TBH I am unsure what this, looking for the encoders, was ever implemented into the driver...
But like I say I am dumb when it comes to coding, so it may not be that easy... 👍

xthestreams, 2020-06-12:
Here's the error for reference.

[image]

AstroOlly, 2020-06-12:
Hmmm, autohome not tested.... 🤔🤔

geehalel, 2020-06-12:
@AstroOlly  Just try to think how to implement autohome: you start from an unknown position and every encoders (microstep or auxilliary) you have are relative. In which direction do you send RA and DEC axis for searching the home indexers without intercepting the pier ? It seems that the auxilliary encoders are used to compute these directions (remember also that swapping a wire in the stepper motors connection will reverse their directions, so the driver doesn't even know how all that thing rotates in real). Now I'm not sure how it really works... And the warning was there because I had never tested the autohome algorithm in real when I put it on github. Now Jasem and other people reported that it works, it may be suppressed.

I've looked at the EQ8-R/RH manual (I find it strange that skywatcher don't put these mounts in their products list) and something caught my eye: "Hard Stops". Does anybody have information about that feature ? One way to make hard stops could be to use accelerometers to get a rough idea of how are both axis relative to ground (and the latitude settings too). A close-up picture of the mount boards may help here. But that's just an idea that crossed my mind.

AstroOlly, 2020-06-12:
@geehalel 
I have my encoders turned off all the time, I never use them and never will as they are pretty rubbish on the original EQ8, yet my autohome works fine.
As for hard stops, there are physical pegs fitted in the mount on the RA axis, so that axis will only turn about 240 degrees, 120 each side of home position, this is to stop cable wrap issues inside the mount...this is in all EQ8 mounts...
I still don’t believe that the autohome function uses the encoders at all, because if it did, then how come it’s works on the new EQ8-r using the skywatcher synscan handset, and yet that mount  has no encoders fitted at all...?

geehalel, 2020-06-12:
Well I took the time to look at the code and you're right, @AstroOlly , the algorithm doesn't use aux encoders. They are just turned off at the beginning. The error comes from a mount type test in some indexer functions. That means that EQ8-R and EQ8-RH have different mount codes than the old EQ8. May you send the mount code so I add it to the known mount list ?
By the way you may test a modified version of the indi-eqmod driver from the [autohome branch of my forked repo](https://github.com/geehalel/indi-3rdparty/tree/autohome). Beware that I am absolutely **not sure of how it will behave**, so perform the test without a scope.
The question remains that I still don't know what the read indexer functions return, the algorithm resets, gets, and gets last read values and is able to deduce directions with that. And I don't have such a mount to know the real values which are returned while the algorithm runs.
Thanks for the info about hard stops, that's so simply evident.

AstroOlly, 2020-06-12:
@geehalel 
Well, I have no idea where to get the mount codes, what exactly are they...?
Like I say I am as far from a software coder as you can get.... 😂
Maybe @xthestreams  can help with that....

xthestreams, 2020-06-13:
Is there an API call I can embed into the forked branch that will return the ID or a log I can take a look at?

I assume I’ll need to compile this on my RPi that’s currently the mount controller, or is it better if I compile and test on my Mac with direct connect to the mount?

geehalel, 2020-06-13:
The mount code is queried by the driver in the very first message (":e"), you may see it if you enable debug (serial comm.). It seems that kstars does not display anymore debug messages so I debug to a log file and here is what it looks like:
<code>
COMM	180.063998 sec	: dispatch_command: ":e1", 4 bytes written
COMM	180.064063 sec	: read_eqmod: "=030A04", 8 bytes read
</code>
The mount code is in the answer "=030A04", it's the last byte ("04"), the first two bytes being the firmware version.
You may compile on the RPi and, if that works, you could keep it until the pull request. Anyway stay near the mount when testing.

xthestreams, 2020-06-13:
Gotcha, I might need to check as to where to install it or will the makefile take care of it?

I’ll log another EQMod bug for the PPEC feature, it’s broken for the EQ6-R and EQ8-R, causes a crash when enabled if training hasn’t already been done. It’s also broken on the EQ8r when training has enabled, but I’m not sure if that’s a me bug, mount bug or EQMod bug.

geehalel, 2020-06-13:
You have to run from an empty build directory
<code>
cmake -DCMAKE_INSTALL_PREFIX=/usr      /path-to-modded-indi-eqmod-git-clone
</code>
<code>
make
</code>
<code>
make install # as root
</code>
assuming indi is installed in <code>/usr</code>. It will replace your current indi-eqmod driver.
Concerning PPEC, create the issue in the  [indi-3rdparty github repo](https://github.com/indilib/indi-3rdparty) as indi-eqmod is a 3rd party driver.

AstroOlly, 2020-06-13:
I just hope none of this ***** up the existing driver that works ok for us old EQ8 users... 😂😂

xthestreams, 2020-06-14:
haven't been able to get this done today - i got distracted trying to work out how to enable cut and paste on my Stellarmate

xthestreams, 2020-06-16:
managed to pull a log file of the interaction - hopefully this help;

[2020-06-16T12:13:58.942 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":j1\", 0 bytes written "
[2020-06-16T12:13:58.943 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=000080\", 8 bytes read "
[2020-06-16T12:13:58.943 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":j2\", 0 bytes written "
[2020-06-16T12:13:58.946 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=007BAA\", 8 bytes read "
[2020-06-16T12:13:58.946 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[SCOPE] Current encoders RA=8388608 DE=11172608 "
[2020-06-16T12:13:58.946 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":f1\", 0 bytes written "
[2020-06-16T12:13:58.956 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=301\", 5 bytes read "
[2020-06-16T12:13:58.956 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":f2\", 0 bytes written "
[2020-06-16T12:13:59.227 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=301\", 5 bytes read "
[2020-06-16T12:14:00.043 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[SCOPE] Compute local time: lst=5.54618658 ( 5:32:46.27) - julian date=2459016.59305004 "
[2020-06-16T12:14:00.044 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":j1\", 0 bytes written "
[2020-06-16T12:14:00.044 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=000080\", 8 bytes read "
[2020-06-16T12:14:00.044 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":j2\", 0 bytes written "
[2020-06-16T12:14:00.045 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=007BAA\", 8 bytes read "
[2020-06-16T12:14:00.045 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[SCOPE] Current encoders RA=8388608 DE=11172608 "
[2020-06-16T12:14:00.045 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":f1\", 0 bytes written "
[2020-06-16T12:14:00.045 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=301\", 5 bytes read "
[2020-06-16T12:14:00.045 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":f2\", 0 bytes written "
[2020-06-16T12:14:00.046 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=301\", 5 bytes read "
[2020-06-16T12:14:00.446 AEST INFO ][           org.kde.kstars.indi] - EQMod Mount :  "[WARNING] *** AutoHome NOT TESTED. Press PERFORM AGAIN TO CONFIRM. *** "
[2020-06-16T12:14:00.965 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[SCOPE] Compute local time: lst=5.54646687 ( 5:32:47.28) - julian date=2459016.59306169 "
[2020-06-16T12:14:00.965 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":j1\", 0 bytes written "
[2020-06-16T12:14:00.965 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=000080\", 8 bytes read "
[2020-06-16T12:14:00.966 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":j2\", 0 bytes written "
[2020-06-16T12:14:00.966 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=007BAA\", 8 bytes read "
[2020-06-16T12:14:00.966 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[SCOPE] Current encoders RA=8388608 DE=11172608 "
[2020-06-16T12:14:00.966 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":f1\", 0 bytes written "
[2020-06-16T12:14:00.966 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=301\", 5 bytes read "
[2020-06-16T12:14:00.966 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":f2\", 0 bytes written "
[2020-06-16T12:14:00.966 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=301\", 5 bytes read "
[2020-06-16T12:14:01.182 AEST INFO ][           org.kde.kstars.indi] - EQMod Mount :  "[INFO] Starting Autohome. "
[2020-06-16T12:14:01.192 AEST INFO ][           org.kde.kstars.indi] - EQMod Mount :  "[INFO] AutoHome phase 1: turning off aux encoders "
[2020-06-16T12:14:01.193 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[MOUNT] SetFeature() : Axis = 1 -- command=5 "
[2020-06-16T12:14:01.193 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":W1050000\", 0 bytes written "
[2020-06-16T12:14:01.193 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=\", 2 bytes read "
[2020-06-16T12:14:01.193 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[MOUNT] SetFeature() : Axis = 2 -- command=5 "
[2020-06-16T12:14:01.193 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":W2050000\", 0 bytes written "
[2020-06-16T12:14:01.193 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=\", 2 bytes read "
[2020-06-16T12:14:01.194 AEST INFO ][           org.kde.kstars.indi] - EQMod Mount :  "[INFO] AutoHome phase 1: resetting home position indexes "
[... cut ...]
[2020-06-16T12:14:01.962 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=000080\", 8 bytes read "
[2020-06-16T12:14:01.962 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] dispatch_command: \":j2\", 0 bytes written "
[2020-06-16T12:14:01.968 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=007BAA\", 8 bytes read "
[2020-06-16T12:14:27.181 AEST DEBG ][           org.kde.kstars.indi] - EQMod Mount : "[COMM] read_eqmod: \"=000080\", 8 bytes read "

geehalel, 2020-06-16:
Thanks, it seems you still run the official indi-eqmod driver, thus the autohome aborts because of the mount code checking.
To get this mount code I need the very first COMM messages when you connect your mount with the 'Connect' switch. You may disconnect immediately thereafter and gets the log file at that time.

AstroOlly, 2020-06-16:
Looks like you two are making some progress on this... 👍😀

xthestreams, 2020-06-16:
@geehalel I was blessed to have Jasem join me on a debugging run tonight and the code fix has generated the desired result and the mount is now Autohoming correctly. 

I will attempt some bounds checking just to ensure I can see that it doesn't do anything unexpected, but I think you can safely commit the change that Jasem tested.

xthestreams, 2020-06-24:
one week later and it seems to be working great

closed 2020-06-25 by knro

geehalel, 2020-06-27:
@xthestreams thanks for performing tests and reporting that it works. And a big thank to @knro for helping debug that stuff. 
It would really be great if I could add the EQ8-R mount code in the driver, I'm not sure if allowing autohome for every mounts would be safe. The EQ8-RH has an absolute encoder in RA and I wonder if that algorithm may work.
May you enable debug comm just before you connect the mount, then just connect and disconnect and post the log file here ?
I'll make the pull request afterwhile.
Thanks.

xthestreams, 2020-06-28:
@geehalel - ah I see, so right now it doesn't just skip the check for encoders on the 8-R it just assumes any mount can Autohome - interesting idea! I would like to try it with a EQ6 or 6-R to see what happens, I don't have mine any more, let me see if I can find one.
Let's connect offline and see what we can do, my EQ8-R is currently dead due to horrible reverse polarity power supply problem, will not be live again for a few weeks.
@knro if we're not releasing this to the trunk then we might want to keep it open until we've got something more solid. The mount command documentation doesn't give any ideas?

AstroOlly, 2020-06-29:
It won’t work on any other SW mounts other than EQ8, as they don’t  have the homing sensors on either of the axis...

xthestreams, 2020-08-21:
@geehalel - after two months of waiting for a new board, the mount is on it's way home shortly - I will be Abel to test again soon. What logs would you like?

geehalel, 2020-08-30:
@xthestreams I just need the log when you connect the mount: simply connect and disconnect the mount then get the log file. 
Tell me also if you keep your indi core library in sync with github, or if you still have the same version as before. In the first case, I would have to also sync my repository.

seescho, 2020-11-22:
I just tried my new EQ8-R with the autohome-function. No success. Incorrect mount type.
How can I help to solve this issue?
Edgar

xthestreams, 2020-12-04:
Posting @seescho solution to the problem - mount code 0x20