indilib/indi #1168: LX200Telescope Guide commands issued during a slew

amendolajamie, 2020-06-08:
**Describe the bug**
The issue is with sending guide commands during a slew in drivers/telescope/lx200telescope.cpp

I have discovered the cause and have a fix. I've never contributed to the indi source, so I'm not sure how to do that. The issue is with sending guide commands during a slew. If that ever happens then the mount goes haywire. The simple fix is to return an error within each of the Guide<direction> calls in drivers/telescope/lx200telescope.cpp

Like this:

IPState LX200Telescope::GuideNorth(uint32_t ms)
{

if(TrackState == SCOPE_SLEWING){
return IPS_ALERT;
}

I tested this last night and everything is working perfectly now. It never does a wild slew. I don't think that there would be any side effects to this, but I'm not sure. I can't imagine a use case where a guide command should happen during a slew.

https://indilib.org/forum/mounts/7065-losmandy-slew-goes-haywire.html

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run indi / ekos / phd2
2. Start guiding with phd2
3. Issue slew command

**Expected behavior**
Mount slews to new position, and continues guiding

**Screenshots**
Don't have any

**Desktop (please complete the following information):**
Raspberry pi 4 
Software compiled on PI:
Indi: HEAD detached at v1.8.5
indi-3rdparty: Indi: HEAD detached at v1.8.5
kstars: HEAD detached at v2.9.8
Phd2: HEAD detached at v2.6.8

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

amendolajamie, 2020-06-08:
I'm not sure how to attach logs. There are in the forum link.

knro, 2020-06-09:
Thank you for the report. Did you find out how are the guiding pulses are sent during a **slew** ? Usually guiding starts when the mount is **tracking**.

amendolajamie, 2020-06-09:
I start guiding using phd2, that's how it is started. At this point I believe that the mount is tracking. Then I issue a slew command.

knro, 2020-06-09:
Alright, I think this is safe to include, if not strictly necessary for other LX200-based mounts.

commit 6443ad3c (2020-06-09): Reject guiding if slewing or parking is in progress.

Fixes #1168
closed 2020-06-09 by knro