indilib/indi #1170: Pegasus Powerbox Advance - No connection

jamiecflinn, 2020-06-11:
**Describe the bug**
New Powerbox Advance - test on Windows OK.  Connect to Linux current Ubuntu running latest Kstars/Eko/INDI - all devices connect to the unit are operable but the Pegasus Unit itself cannot connect - prevoius model works correctly
notice the returncode of PPBA_OK - I suspect that the code is looking for PPB_OK and that the change in name is the reason
Have tried multiple cables - no change

**To Reproduce**
Exact steps to reproduce the behavior.
1. Select Pocket PowerBox in profile setup
2. Connect all device to the unit - run Kstarrs/EKOS - Monitor INDI control panel and debug messages - no connection on any port
[2020-06-10T21:25:56.751 EDT DEBG ][           org.kde.kstars.indi] - Pegasus PPB : "[DEBUG] Connecting to /dev/ttyUSB1 @ 9600 "
[2020-06-10T21:25:56.751 EDT DEBG ][           org.kde.kstars.indi] - Pegasus PPB : "[DEBUG] Port FD 3 "
[2020-06-10T21:25:56.751 EDT DEBG ][           org.kde.kstars.indi] - Pegasus PPB : "[DEBUG] Connection successful, attempting handshake... "
[2020-06-10T21:25:56.751 EDT DEBG ][           org.kde.kstars.indi] - Pegasus PPB : "[DEBUG] CMD <P#> "
**[2020-06-10T21:25:56.778 EDT DEBG ][           org.kde.kstars.indi] - Pegasus PPB : "[DEBUG] RES <PPBA_OK> "**
[2020-06-10T21:25:56.778 EDT DEBG ][           org.kde.kstars.indi] - Pegasus PPB : "[DEBUG] Handshake failed. "
[2020-06-10T21:25:56.783 EDT INFO ][           org.kde.kstars.indi] - Pegasus PPB :  "[WARNING] Communication with /dev/ttyUSB1 @ 9600 failed. Starting Auto Search... "

**Expected behavior**
connectoin as per older model

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04.4 LTS
 - 

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html
[log_21-35-55.txt](https://github.com/indilib/indi/files/4762057/log_21-35-55.txt)

jamiecflinn, 2020-06-11:
Look like i may be correct:
 LOGF_DEBUG("RES <%s>", response);

    setupComplete = false;

    return (!strcmp(response, "PPB_OK"));

the new unit is returning PPBA_OK so this will fail....

knro, 2020-06-11:
If you change that to PPBA_OK, does everything else work as is? What's the difference between this and PPB? i.e. are they identical or have differences? If yes, what are they?

jamiecflinn, 2020-06-11:
This is the only difference I know of and do not have an environment to build and test at this time (best I can do right now is read the code) - according to Pegasus website they have this unit running on INDI which makes me think they hacked their own FW to respond with PBB_OK - if you can advise on quick environment setup I can do a build and test by dropping in a new dll (never did understand Linux installs)

jamiecflinn, 2020-06-11:
[log_09-28-57.txt](https://github.com/indilib/indi/files/4765045/log_09-28-57.txt)
Took a while to get environment up for building but i have now run tests and simply changing PPB_OK to PPBA_OK  brings everything on line....I am not familiar with GitHub pull request so not ure if I shold make real changes to the drive code or you want to do it - I have NOT looked at other new functions like focuser on tis unit to see if they show up, but being able to get dew control and temp is good enough for me - I guess we can close thi but at some point either a new driver or updated single driver for original and advance units will be needed....Thanks Jasem!...

closed 2020-06-11 by knro

inglis-dl, 2020-06-12:
I am also hoping PPBA driver will work on linux.  There a hardware differences between PPB (no usb hub) and PPBA (4 port usb 3 hub).  The control set is not published on Pegasus Astro site for PBBA but is available for PPB and PPBU.  Here is what Evans at Pegasus astro emailed me on this issue:
"It has pretty much the same command set with some extended commands to support new functionality.
I think the problem with PPB Advance indilib driver is the response when driver connects.
In Pocket Powerbox it receives PPB but in Advance the response is PPBA
Example for Pocket Powerbox :
Transmit: PA 
Receive: PPB:12.2:0.5.22.2:45:17.2:1:1:120:130:1
"

jamiecflinn, 2020-06-12:
Hi…

Tested last night and was successful – this was on an Ubuntu 18.04.x system  - all ports are OK and all connected devices operate – dew system was OK and were temp sensors – the only problem after changing the handshake was that the #1 USB3 port is ONLY USB3 and cameras that require USB2 discovery are never found – I have raised tgis with QHY as to why they need a USB3 port that supports USB2 to RUN in USB3 mode.

The Drivers in INDI will need to be changed though as the PPBA now provides focus control and some other abilities….for now however if someone can make a permanent change to the handshake code to look for PPB_OK and PPBA_OK it will be universally usable with the existing PPB driver(modified)

knro, 2020-06-13:
So looks like it will partially work  now but needs to support the extra features. Anyone willing to add these to the driver?

inglis-dl, 2020-06-13:
I checked out the code and started pegasus_pba.h and pegasus_pba.cpp,
a hybrid of ppb ahd upb

inglis-dl, 2020-06-14:
I think the ultimate power box (UPB) feature set and command table seems more suited to the power box advance (PBA) than the PPB... have reached out to Pegasus to post the serial command table specific to PBA

atoy40, 2020-06-17:
Hello

@inglis-dl If you need help to implement the PPBA code, I'm available. I received my PPBA yesterday, and I'm a developer (I'm working on a nodejs wrapper for indi, "atoy40/indijs" on github). Have you received the serial spec from Pegasus ?

And I agree, the PPBA is more a small UPB than a PPB. (usb 2/3 hub, adjustable voltage output, ...)

Thanks
Anthony.

inglis-dl, 2020-06-17:
Hi Anthony,

haven't received it yet but they assured me they would provide it "soon"

best,
Dean

inglis-dl, 2020-06-22:
PPBA serial command table is up on the PA site ... time to start coding!

atoy40, 2020-06-23:
@inglis-dl I started to code a driver last weekend by testing command trough serial, because I needed to change the Adj output voltage for my last astro night :). So yesterday I added some feature following the command table. I also fixed some bugs in the original PPB driver.
Here is a screen shot with a indi client I'm working on :
[image]

I've forked the project and I'll push my modification today on my repo.

Anthony.

inglis-dl, 2020-06-23:
nicely done! you beat me to it.   I have the full serial command table from
pegasus astro specifically for PPBA.
Ill give your code a try.

inglis-dl, 2020-06-23:
see attached.

atoy40, 2020-06-23:
Pegasus published it on the website yesterday.
Here is a first version (added LED on/off and some new current/consumption stats compare to the screenshot) :
https://github.com/indilib/indi/pull/1185

Anthony.

commit ccc433f5 (2020-06-11): Add PPBA which should fix #1170