indilib/indi #1173: Issues with Planewave EFA driver

TheVirtualTim, 2020-06-13:
**Describe the bug**

Multiple issues experienced using KStars/Ekos with PlaneWave EFA focus driver using StellarMate OS.  

Issue A:

Driver seems to be able to focus 'out', but has problems when trying to focus 'in'.  Sometimes it refuses to move at all.  Other times it moves a fraction of the expected distance and declares it has 'reached requested position'.

Issue B:

Maximum position steps seems to default to a different model Hedrick focuser.  PlaneWave makes this focuser for their 12.5, 14, 17, 20, and 24" scopes.  Focus limits are likely different for each model.

When I attempt to enter the maximum position (steps) for my model, it refuses to let me enter the full value.  I am attempting to enter 6218628 but am unable to type the final digit (8 in my case).

The following info is from PlaneWave's own driver for the EFA used with Hedrick focuser on on the CDK 12.5" telescope model:

Conversion: 115.134 Encoder/Microns
Max speed: 100000 Encoder/sec
Backlash: 0
Go to Slow Range: 150000 encoder
Range limit max: 6218628 encoder
Range limit min: 0
Home: 0

It appears the planewave_efa driver wants to use 'steps' (I think that's what PlaneWave refers to as encoder positions).  They do offer the conversion factor (possibly different for each model).  On my model it is 115.134 encoder steps per micron.

Issue C:

Secondary mirror temp is reported as -100 (it should read very close to ambient since the dew heaters are not running).  This scope is equipped with the Delta-T dew heater.  That driver is also installed and ... with quick/casual testing it does appear to be working although I only told it to turn the heaters on briefly and then off again.  But the Delta-T driver does seem to display the temperatures.

Issue D:

I frequently get the message:  [ERROR] Invalid checksum!  (in the message window)

**To Reproduce**
Exact steps to reproduce the behavior.
1. Start KStars/Ekos and configure a profile to include at least the planewave_efa focuser driver and planewave_DeltaT dew-heater driver.
2. Start the INDI server, configure the serial ports (in my case /dev/ttyUSB0 for the EFA and /dev/ttyACM1 for the DeltaT (the Losmandy Gemini II mount is using /dev/ttyACM0).  Also 'connect' each device and verify connection succeeded.
3. Scope was already in encoder position 0 (all the way 'in').  Set the sync value to '0' and set 'Calibrated' state.
4.  Set focus direction to 'Focus Out'
5.  Enter a relative position (e.g. 100000 steps) and hit 'set'

Focuser will gently focus outward and display that it has reached the desired position (this seems to work correctly).

6.  Enter '0' as the absolute position and press 'set' to focus back in.
7.  Focuser briefly moves inward, stops, and reports it has 'reached requested position' although it's clear that it hasn't moved enough.  Also, the 'absolute' position wont read a value near zero.  It seems to make it roughly the low 90,000's and stops here (it varies each time).

8.  At this point even if I use the hand-paddle with the EFA it refuses to focus inward any further (I can focus 'out' and then focus back 'in' to the previous point, but cannot go beyond it.  

9.  If I set the 'sync' position to a higher value (such as 200,000) and 'set', then I can try to focus inward again and it will make a little more progress.  I have to repeat this many times to get the focus to travel all the way in.

Focuser is mechanically fine (it is not damaged).  If I disconnect the the Raspberry Pi / StellarMate OS and attach a Windows PC running the PlaneWave software, the focuser works as expected.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS:  Raspberry Pi 4B (4GB) running Ubuntu 18.04.4 LTS  (StellarMateOS image)

 - Version 
Originally StellarMateOS 1.4.2 but this has been updated (apt-get update & upgrade) ... it should be at 1.5.2 but I'm uncertain where the version info is recorded.

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html
[log_16-37-08.txt](https://github.com/indilib/indi/files/4775474/log_16-37-08.txt)

knro, 2020-06-14:
Thank you for the report. I think you're using a slightly older version. I suggest you install v1.5.2 (you cannot upgrade from 1.4.x to 1.5.x) and then switch to nightly/beta channel and use the latest driver and report back.

TheVirtualTim, 2020-06-15:
I installed 1.5.2 and updated it... but the updates is much different than in 1.4.2 (my old version).  The SM updater seems to have a button for "Stable" which can be highlighted or not.  I've clicked it so it isn't highlighted.  Is there something I can check to confirm it's on the nightly/beta channel?

I did notice the secondary mirror temp is no longer reported (it was erroneously reported as -100).

Other issues still persist... 

A - after focusing 'out' it wont let me focus all the way back 'in' (stops in the low 90,000's range)
B - can't enter max steps (field wont let me enter the last digit)
D - still get occasional checksum errors

knro, 2020-06-16:
Ok I need TeamViewer access to check it out. Send it to support@stellarmate.com

knro, 2020-07-01:
I believe this issue is resolved, please confirm.

TheVirtualTim, 2020-07-02:
Hi Jasem,

It still seems that it is not resolved.

I tried to test the focuser today — still not working.  I noticed the maximum steps are set to the old value and tried to update it to the correct value (which is 6218628 encoder steps).  The software still wont let me enter a value that large.

I recall you mentioned you updated it to allow for a value up to 10 million steps — which I cannot enter.

I’m wondering… am I use a different driver than you were testing?  (e.g. if there a test driver in a different directory)?

I ask because I noticed you create a new profile with all simulators *except* for the PlaneWave EFA but it was set to use a ‘remote’ INDI server (which happened to be on ‘localhost’ rather than just using the local option.  This made me wonder if you were starting a different INDI server instance than the one that KStars/Ekos starts.

Possibly I just need to copy a driver to the correct directory to test??

Thanks & Best Regards,
Tim

knro, 2020-08-30:
Hello Tim, is this still an issue? All the drivers should be up to date now.

comptonizing, 2020-09-19:
Dear Tim, Jasem,
we have a similar problem with the EFA kit on our observatories CDK20, but in our case the connection always failed allegedly due to an invalid checksum, so I started looking into this problem myself. It quickly turned that the checksums are calculated correclty (i.e. 0xc4) but the checksum sent along with the message on the bus was 0xffffffc4 which got us thinking about the sign of the chars. Indeed, when interpreting the data sent by the EFA as unsigned chars everything is working as expected, see the pul request above. This can also explain Issue A and Issue B reported by Tim with random overflows in the signed chars during the reconstruction of the actual integer.
However, now two minor problems are still showing up:
1. I get exactly one timeout on the TTY everytime I connect to the EFA when the driver is attempting to read the calibration state
2. When using the hand control to focus while the indi driver is connected it throws a lot if invalid checksum errors, but all operation continues just fine regardless
Ciao,
Philipp

knro, 2020-09-19:
Thank you Philipp for the feedback. The checksum issue should be easy to fix. Let me try to do that and then check how it affects the other issues.

knro, 2020-09-19:
So the checksum formula used by Planewave is:
```
for (uint32_t i = 1; i < len - 1; i++)
        sum += cmd[i];
    return (-sum & 0xFF);
```

Is the issue in the return where the sum is negated? or in the summation itself?

closed 2020-09-28 by knro