indilib/indi #1176: Enabling PPEC without training first causes EQMod to crash

xthestreams, 2020-06-16:
**Describe the bug**
When using an EQ6-R or EQ8-R if PPEC for either RA or DEC is switched on AND the mount has not yet been trained, the driver crashes and the kstars session crashes requiring a reconnect.
 
**To Reproduce**
1. Find a PEC capable mount (tested with EQ6-R/EQ8-R) that is not PEC trained or full factory reset (eg: motor firmware upgrade)
2. Click PPEC On to enable in the control panel.
3. The driver appears to crash causing KStars to either crash or to terminate session

**Expected behavior**
It should ideally check with the motor to find out if it has been trained and either be grayed out or issue a message to "train before enabling"

**Screenshots**
N/A - I can provide a video recording if helpful.

**Desktop (please complete the following information):**
 - OS: RPi with Mac Client
 - Version not sure, but has been consistent over the last 3 months including a full upgrade of Stellarmate

**Log Files**
I am having trouble finding any logs, where are they stored?How can I enable?

knro, 2020-06-16:
Thank you for the report. Can you provide the GDB backtrace?

xthestreams, 2020-06-16:
can do - where would I find it?

knro, 2020-06-16:
Are you using StellarMate or something else? instructions vary.

xthestreams, 2020-06-16:
Stellarmate of course! gotta support open source!

xthestreams, 2020-06-24:
quick update - knro used TV to review the bug live and I believe understands the source of the problem

xthestreams, 2020-08-21:
When the mount returns I will re-run this test. Once I understand what's going on will take some time to document how PPEC training works for this mount with INDI so that people are doing it right - I suspect I was doing it all wrong to make things worse.

knro, 2020-12-09:
Was this issue fixed?

xthestreams, 2020-12-09:
Hey Jassem! I'll be back tomorrow - I somehow lost TV access to the Stellarmate box while travelling, as soon as I am home I'll test and let you know.

xthestreams, 2020-12-10:
Confirmed - still occurring.

xthestreams, 2021-01-06:
@knro - i am about to replace this mount, now that I have it back I remember all the things about it I don't really like! Before I sell it let me know if you want me to place it onto its own INDI server to run tests on.

knro, 2021-09-28:
Closing here since this is the 3rd party driver issue.

closed 2021-09-28 by knro