indilib/indi #1181: Let AZ-GTI driver render the mount's SNAP port available to the Gphoto driver to take >30sec pics with older DSRLs

federicoparra, 2020-06-19:
When using a DSLR that requires a separate shutter-trigger cable for >30sec pictures in Gphoto, and simultaneously using AZ-GTI mount, gPhoto should "see" the AZ-GTI's SNAP port. NIKON and CANON produce off-the-shelf, very cheap SNAP compatible cables (the AZ-GTI actually comes with a free SNAP cable that works with some DSLR directly). The Synscan app allows to use the SNAP port to trigger pictures of any desired exposure duration with camera in bulb mode.

geehalel, 2020-06-27:
Hi,
Are you using the AZ-GTI driver based on indi-eqmod ?
If this is the case, I would need the mount code to manually add it in the driver as the number of snap ports is not available from the firmware. If you enable debug Serial comm before connecting the mount in the control panel, then just connect and disconnect, the log file will contain the desired code.  I believe I will make this code visible in the firmware tab in next release.

federicoparra, 2020-07-01:
Perfect, I'll do this as soon as I get the chance, hopefully this weekend. thank you!

federicoparra, 2020-07-12:
> Hi,
> Are you using the AZ-GTI driver based on indi-eqmod ?
> If this is the case, I would need the mount code to manually add it in the driver as the number of snap ports is not available from the [...]

Hi! I tried today but couldn't find the option "enable debug serial comm" anywhere. Could you guide me a little bit to find it? Thank you! Federico

geehalel, 2020-07-13:
Hi,
Before connecting the mount in the control panel,  go to the "Options" tab and "Enable" "Debug".
You should see other options appearing at the bottom of the tab, check 'Serial Port' in the "Logging Levels" property and
"To Log File" in the "Log Output" property. When you check this one, you should see an info message stating where is the log file:
```
2020-07-13T18:47:40: [INFO] Session log file /home/xxx/.indi/logs/2020-07-13/indi_eqmod_telescope/indi_eqmod_telescope_18:35:36.log 
```
Then just connect the mount in the "Main control" tab, and when it's done (2 seconds) disconnect the mount.
Attach the log file here, or get the line at the beginning of the file containing the first sent command:
```
COMM	119.124737 sec	: dispatch_command: ":e1", 4 bytes written
COMM	119.124784 sec	: read_eqmod: "=020300", 8 bytes read
```
Thanks.

federicoparra, 2020-07-15:
Strangely, both using INDI control panel on the Raspberry directly or on Windows connected to the Raspberry server, when I click on Enable debug for this specific driver it very quickly turns off again; this is my trying repeatedly to enable it:
2020-07-15T13:09:38: [INFO] Debug is disabled. 
2020-07-15T13:09:38: [INFO] Debug is enabled. 
2020-07-15T13:09:37: [INFO] Debug is disabled. 
2020-07-15T13:09:37: [INFO] Debug is enabled. 
2020-07-15T13:09:36: [INFO] Debug is disabled. 
2020-07-15T13:09:36: [INFO] Debug is enabled. 
2020-07-15T13:09:35: [INFO] Debug is disabled. 
2020-07-15T13:09:35: [INFO] Debug is enabled. 
2020-07-15T13:09:34: [INFO] Debug is disabled. 
2020-07-15T13:09:34: [INFO] Debug is enabled. 
2020-07-15T13:09:32: [INFO] Debug is disabled. 
2020-07-15T13:09:32: [INFO] Debug is enabled. 

Any ideas?

federicoparra, 2020-07-15:
> Hi,
> Before connecting the mount in the control panel, go to the "Options" tab and "Enable" "Debug".
> You should see other options appearing at the bottom of the tab, check 'Serial Port' in the [...]

Strangely, both using INDI control panel on the Raspberry directly or on Windows connected to the Raspberry server, when I click on Enable debug for this specific driver it very quickly turns off again; this is my trying repeatedly to enable it:
2020-07-15T13:09:38: [INFO] Debug is disabled.
2020-07-15T13:09:38: [INFO] Debug is enabled.
2020-07-15T13:09:37: [INFO] Debug is disabled.
2020-07-15T13:09:37: [INFO] Debug is enabled.
2020-07-15T13:09:36: [INFO] Debug is disabled.
2020-07-15T13:09:36: [INFO] Debug is enabled.
2020-07-15T13:09:35: [INFO] Debug is disabled.
2020-07-15T13:09:35: [INFO] Debug is enabled.
2020-07-15T13:09:34: [INFO] Debug is disabled.
2020-07-15T13:09:34: [INFO] Debug is enabled.
2020-07-15T13:09:32: [INFO] Debug is disabled.
2020-07-15T13:09:32: [INFO] Debug is enabled.

Any ideas?

gubihu, 2020-07-18:
You also have to enable debug in Kstars settings, otherwise it is EKOS that turns debug off. I had troubles with that too when I tried to submit my first log.

federicoparra, 2020-07-22:
[log_18-41-35.txt](https://github.com/indilib/indi/files/4961335/log_18-41-35.txt)
Hope it has what you need!

geehalel, 2020-07-23:
Thanks @gubihu, I did not think that Ekos may turn debugging off automatically.
@federicoparra yeah that's it, mount code is A5.
I currently have some failures with my development computer, so won't be able to make changes to my repo before some days.
Would you be able to compile and test the modified driver before I make a pull request ?

federicoparra, 2020-07-23:
> Would you be able to compile and test the modified driver before I make a pull request ?

You mean once you do the changes? Yes, I can try it myself.

geehalel, 2020-07-26:
Ok, I did the changes in my repo in the [azgti-snap branch](https://github.com/geehalel/indi-3rdparty/tree/azgti-snap).
I tested with the latest indi-core github version, there may be some errors in kstars when disconnecting (dispatch command errors),
it seems drivers do not need to delete properties anymore.

federicoparra, 2020-09-09:
> Ok, I did the changes in my repo in the [azgti-snap branch](https://github.com/geehalel/indi-3rdparty/tree/azgti-snap).
> I tested with the latest indi-core github version, there may be some errors in [...]

Hi! I have some time in my hands (on vacations) and have my AZ-GTI :) so how do I test this? I have the SNAP cable with me as well :)

geehalel, 2020-09-10:
You would have to compile the indi-eqmod driver of my repo branch, it should work with the latest indi core (I just tested, still dispatch command errors on disconnection). There will be a "Snap Port 1" switch in the Options tab when you run the `indi_azgti_telescope` driver and connect your mount. If you plug the SNAP cable to your DSLR, you will be able to toggle bulb exposure using this switch.

knro, 2020-09-27:
@geehalel is this ready to be merged with master at this stage? Thanks!!

geehalel, 2020-09-27:
I'm still waiting someone to test on a **real** AZGti mount, I just tested with the simulator using the A5 mount code, I can not be sure that it toggles whatever on a real mount...

federicoparra, 2020-09-27:
@geehalel @knro I'm a newby with Linux, only using it with Stellarmate, but I do have the AZ-GTI, if any of you let me know the steps I'll try it asap. Thank you so much for your work!

geehalel, 2020-09-27:
Ok I will do a pull request after syncing the branch to the latest commit, so you may be able to test in your stellarmate setup when it will be updated (I don't know when this will happen however). The main problem I can see is that the snap port on the AZGTi is not tied to the first output but to the second one. In order to test, connect your DSLR in Manual/Bulb mode to the snap port and try starting an exposure by pressing the Connect button in the Snap port control of the EQMod Indi panel. And Disconnect to stop the exposure.

geehalel, 2020-10-05:
Pull request just done. Wait until it comes to your stellarmate (you should see your mount name AZGTi now in the firmware tab, and the snap port control button in the Options tab).
Please confirm that works as expected.
Thanks.

federicoparra, 2020-10-05:
> Pull request just done. Wait until it comes to your stellarmate (you should see your mount name AZGTi now in the firmware tab, and the snap port control button in the Options tab).
> Please confirm that works as expected.
> Thanks.

Perfect, I'll try asap

federicoparra, 2020-10-07:
@geehalel @knro I updated Stellarmate stable channel but within the EQMod and/or the AZGTI drivers' indi panel there is no Snap port control option. I imagine this is because the latest stable update still does not contain your modification? I'm downloading now the Beta channel's last version to see if the option appears.

federicoparra, 2020-10-07:
Addendum: just tried the beta channel's last update and same thing: no Snap port control option appears.

geehalel, 2020-10-07:
The pull request is still pending, it is in the [indi/indi-3rdparty](https://github.com/indilib/indi-3rdparty/pulls) repository named AZGTi Snap. You have to wait Jasem or some repo admin merges it into the master branch. Then it will have to be included in the Stellarmate updates. And then you will be able to see the Snap option...

federicoparra, 2020-10-07:
@knro could you accept his pull request so that it's included in Stellarmate updates?

federicoparra, 2020-10-20:
@knro  any news?

closed 2021-04-17 by knro

killercode, 2022-03-20:
Hi, are we sure that this was merged?

killercode, 2022-03-20:
Because I am using StellarMate 1.7.0 and there is no change in there :(