indilib/indi #1183: Culmination offset issue on Astroberry with EKOS

jcastro111, 2020-06-20:
I am reporting great difficulties to use with Astroberry in the EKOS Scheduler the Culmination offset condition. For some target, the function is well implemented, for other, a wrong date shifting is set making the function absolutly not usable. 

If any body have an idea of this problem and have an idea which module or file is impacted, I can as an old propmmer in C,  investigate if I can found this potential bug.

regards.
JC

TallFurryMan, 2020-06-20:
This is the indi bug tracker, so this will be closed soon. But if you want to have a look at date shifts, check the KStars repository on invent.kde.org, and search the culmination restriction in ekos/scheduler.cpp and ekos/schedulerjob.cpp files. However, you should verify your geographical location and the DST setting you use first.

closed 2020-06-20 by knro

jcastro111, 2020-06-24:
Thank for reply. DTS is for Day time setting ( Paris is +2 ? in summer  ). Concerning  Geographical location , it is OK.  Just to know where I could submit this issue so ?
In the Scheduler.cpp and schedulerjob.cpp. All remaining to culmination seems effectively defined according to the Altitude / Moon/ position restriction.