indilib/indi #1187: New INDI driver required to control the focuser from Raspberry PI 4.

oldmar, 2020-06-23:
Hi, I am an old pensioner who has rediscovered the astronomy dreamed of as a child and a new user with very limited knowledge of the programming language.
I own a telescope with a focuser to which I have applied an automatism taken from a project found on the net, it works very well but unfortunately it only works with the ASCOM protocol.
My mount has weight limits caused by a heavy mini PC used to control everything, I would like to use the Raspberry Pi 4, I read it works well with the INDI - Kstars - Ekos protocol.
My problem I don't know how to make it work with INDI, I hope there is someone who can make it work and write a driver for INDI
The components I have are:

    Arduino mega 2560
    Arduino Motor Shield
    TRINAMIC QSH4218-51-10-049 STEPPER MOTOR, 1.8DEG, 1A, 0.49NM
    Adafruit MCP9808 temperature control
    Thank you

knro, 2020-06-23:
So this was again created using bug template? Did you use Driver request?

[image]

bluthen, 2020-06-24:
I didn't see which firmware the arduino is running. It is possible it is already supported. What is the open source project?

myfocuserpro2?

oldmar, 2020-06-25:
hi bluthen,
concerning myfocuserpro2 I loaded and tested with my components a version modified by a user found in an Italian forum.
It had compatibility with INDI but it had many problems, stepper current adjustment problem, brake control problem, temperature problem, the stepper was only going to one side, oter setting problems within Ekos, can you tell me how to send you this modified test maybe you could fix it and make it compatible and functional with my components.
Thanks anyway for reading and replying

bluthen, 2020-06-25:
Ahh I see, it is modified version of myfocuserpro2. Sorry I shouldn't of butted in, I thought with the parts you mentioned it was a good chance it was just stock myfocuserpro2 and you didn't know about an existing driver. But you do, it just isn't working, at least for your version.

I looked at using myfocuserpro2 in the past. I had issue with the getting the shock version to work. I didn't like the license, so was not motivated to figure it out and instead took a different focuser firmware and modified it for my particular use-case. Because of that I don't think I personally can look into it.

Good luck.

oldmar, 2020-06-25:
Hi, for me it is not important what the firmware or the project is, using the components that I have priority is that it works with the Raspberry PI 4 and is recognized by INDI - KSTARS -EKOS.
Currently the my focuser works very well with all programs remotely, the problem is that it uses ASCOM and is not compatible with INDI.
If you have a version that works with the Raspberry PI and is recognized by INDI - KSTARS -EKOS I would be grateful if I could make it available so that I can try it out.
Let me know.
Anyway thanks for replying.

Best regards

bluthen, 2020-06-25:
Well there are suppose to be no known issues with the indi driver if using the stock firmware version.
https://indilib.org/devices/focusers/myfocuser-pro-2.html

So you could just try using the latest official version of the firmware, use whatever tools to see you have it working right, and then if it doesn't work it is truly an indi issue with the driver.
https://sourceforge.net/projects/arduinoascomfocuserpro2diy/

So you say your version was modified for your hardware. Which of your hardware did it have to be modified for? I see L298P and Atmega2560 in the official version.

oldmar, 2020-06-25:
as I wrote, the version edited by a user of a forum, had many problems and was not perfectly manageable in INDI-KSTARS-EKOS, some things worked terribly not, like the stepper heated like an iron, only part of it turned on and many other things were wrong ... maybe it's better to abandon that project ... at least for my limited skills with programming languages

oldmar, 2020-06-25:
these are my hardware components
Arduino mega 2560
Arduino Motor Shield
TRINAMIC QSH4218-51-10-049 STEPPER MOTOR, 1.8DEG, 1A, 0.49NM
Adafruit MCP9808 temperature control

bluthen, 2020-06-25:
I modified Hanastro's Focuser code which works through the moonlite protocol. I didn't use the parts listed there as I just wanted to use spare parts I had laying around. So I modified it to use the adafruit motorshield v2 and removed the temperature stuff. Also used a very inexpensive unipolar 32-Step 1/16 gearing stepper motor.

Original: https://github.com/Hansastro/Focuser
Mine: https://github.com/bluthen/Focuser

Not saying it is perfect but it works for me. There are many other open source focuser projects.
https://github.com/glowmouse/beefocus (Also has links to other focuser projects)

If you are interested in something else maybe people at the INDI forum can help:
https://indilib.org/forum.html

oldmar, 2020-06-26:
Hello,
I understood the change you were inspired by, maybe you just observe the beauties of the sky, you don't care about photography, for example having temperature control and the correction of the focuser according to its variation is important, for observation your project is fine but my use is addressed to photography.
You have been very kind and it was a pleasant conversation, in all sincerity, I was hoping for concrete and implementation help for a project that could be applied to my focuser.
However if you can post your modified monolith project, I would like to take a look, maybe I could reinsert the temperature control and other small functions with the help of some programming expert here in this comments ... always better than nothing.
Thanks again for your helpful suggestions ... clear skies.

TallFurryMan, 2020-06-26:
I am not sure I understood the last post clearly, @oldmar. You want a full-fledged focuser free of charge, or you are interested in executing yourself a hand-crafted focuser assembled with random parts?

oldmar, 2020-06-26:
hi,
 @TallFurryMan,
I'm not sure I understand what you mean, I humbly try to interpret your writing and hypothesize an answer.
As you may have read the previous comments, I have no programming skills, I just follow the software track of a free project available on the net, buy the necessary components and assemble them, it has already happened for the automation of my current focuser that works fine ... but only with the ASCOM protocol.
Now, for needs related to the overload limit of my mount used remotely, caused by an excessively heavy mini pc, I would like to try the solution with the use of the Raspberry PI 4 and use the INDI protocol with the good programs that interface with it, Kstars - Ekos ... but I repeat, programming languages ​​for the construction of a driver module for INDI is not in my knowledge. I hope there is some willing altruist who can help me.
If my comments and requests are not compatible in this environment, I understand, apologizing for taking away not space for more appropriate comments, I could remove the trouble,by deleting my subscription 
For my pockets, as you will can see it is not completely free having purchased Arduino mega 2560, Arduino Motor Shield,
TRINAMIC QSH4218-51-10-049 STEPPER MOTOR, 1.8DEG, 1A, 0.49NM, Adafruit MCP9808 temperature control and lastly the Raspberry PI.
I hope I have been able to answer your question, please let me know...
Clear skies

bluthen, 2020-06-26:
Yes, the original project firmware had temperature sensor support but I didn't want to mess with that(at least for now). I use it for astrophotography. I currently just autofocus periodically throughout the night. I do have a bunch of DS18B20 temperature sensors around, eventually I thought I could add that but have not gotten to that yet.

Your electronics looks like it should work with myfocuserpro2 stock firmware, and if it doesn't a bug report can be filed. It wouldn't be a new driver issue, it would be to fix the current myfocuserpro2 driver. There are other people that use myfocuserpro2 with INDI and maybe they can help at the indi forums:
https://indilib.org/forum.html

I'm a little confused about the "if you can post your modified monolith project", do you mean the github repos?
Original: https://github.com/Hansastro/Focuser
Mine: https://github.com/bluthen/Focuser

Do note the original has temperature sensor support. It uses the LM335 for a temperature sensor and A4988 for stepper driver, which should work well. I just didn't have those in my parts bin. Also my stepper was unipolar but the A4988 is strictly for bipolar steppers.

oldmar, 2020-06-26:
hi bluthen,
you are absolutely right to be confused for my write " if you can post your modified monolith project "please excuse me, I had read hastily your writing, in fact it was all specified.
Yes, the myfocuserpro2 driver project is compatible with my components, but it has many problems, you can't imagine how many tests I did, in some cases I risked burning the stepper.
To use the myfocuserpro2 driver project "with my components" and not to risk damage all  it is necessary to know a bit of programming to immediately understand where to act or ask to modify ... unfortunately this is not in my possibilities.
I hope that the use of the Raspberry for INDI involves more amateur astronomers, only in this way will it be possible to find many projects similar to Ascom, or the possibility of combining Ascom and INDI.
Thanks for the support, much appreciated.
Clear skies

oldmar, 2020-06-30:
Greetings to all, please, excuse my written reflection...understand it.
I was hoping to find who could write a driver for my components, I only read general indications, obviously they were appreciated and I thank those who wrote them.
In this environment, my non-preparation in programming is known, it is not a question of wanting to find ready meals.
Maybe I will ask trivial questions and someone will smile, my interest in continuing to use Raspberri PI remains unchanged, so I ask at least for help to know how to write a driver module for INDI, where to start, what to study, whic data you need, in short everything he needs and can serve to create them .
In this way, after learning new knowledge, I can communicate better with programmers.
I hope my request finds a person who shows me an open door, otherwise my presence in this environment makes no sense.
clear skies

TallFurryMan, 2020-06-30:
There should be two parts in your attempt: building the firmware that controls your stepper, and making sure the driver and the firmware understand each other. @bluthen provided you with a firmware, so you need to check the differences in hardware that would have to be resolved before it may be used. Try to enable logging and communicate with the firmware manually using a serial terminal (there are many resources explaining how to do that on arduino wikis). Once you verified all commands are running properly, you can start to use the INDI driver. Don't go too fast and don't expect DIY elements to automagically work with each other.

oldmar, 2020-06-30:
Hello and welcome again @TallFurryMan,
unfortunately the myFocuserPro2 firmware is not compatible with my components, I tried and to be sure I also wrote to the manufacturer of the DY project to confirm it.
I suppose there are two ways left, 1 building new firmware and driver for my components, 2 buying new components, but it would be a waste of money.
From your writing I understood that you need 2 things, 1 frmware - 1 driver,
where can i start to build them myself for my components?
clear skies

TallFurryMan, 2020-06-30:
If you have the firmware source code, I think you can see which libraries it is using. In arduino code, libraries are often named after the hardware they support.

It may be that the library is compatible, but that the wiring is not proper. 

If it is not compatible, find a library that is referred working with your hardware, and replace it in the arduino code. This part is probably better discussed in the arduino forums, it involves downloading the arduino programming IDE, looking at the source code, replacing instructions using the previous libraries with new instructions. It is not overly complicated, but can be confusing at first.

Your problem may not be with the INDI driver but between your firmware and your hardware.

oldmar, 2020-06-30:
I will try the project again following your suggestion, I will reload the project and the libraries in Arduino, then I will send it running with the Raspberry ... with your help let's understand where it doesn't work ...
thank you

TallFurryMan, 2020-06-30:
I think you should not use the Raspberry for your test. You should use the basic serial terminal on your computer and enter commands manually. This way, you will know exactly what part of the firmware requires a rework.

TallFurryMan, 2020-06-30:
This is strange : the driver should be using the protocol commands that you did test. Are you able to relate those commands you used on the terminal with the commands you see in the INDI driver? Perhaps you could recognize a specific and simple use case of the driver that produces the issue, and reproduce it using the terminal to investigate inside the firmware?

oldmar, 2020-06-30:
I'm sorry, my current knowledge does not allow me to understand and do what you have described, the only thing I could always do if you want to find the malfunctions is to send you all the project software including the libraries.
However today following your suggestion to look in depth in github and forum, I read that a gentleman ported the driver for the auto focus INDI that I use in Ascom with my components the link is this ( https: // github. com / Anandir / indi) - (https://github.com/Anandir ) but unfortunately I don't understand how to compile it there are few instructions.

Maybe you with your experience and knowledge can tell me how to install it on my raspberry PI.
You are really kind @TallFurryMan to help me, thank you

bluthen, 2020-07-01:
Ohh.. I think you should make a new issue. Using the "Driver Request" template. Answer all the questions. Make sure to list:
http://www.astropix.it/software/astrofocus/index.html
and
https://github.com/Anandir/indi-astrofocus

oldmar, 2020-07-01:
hello Bluthen, well resentful,

I don't know how to correctly open the request for a new driver "can you help me make the request"? ... I send the same request to @TallFurryMan, your are the only two kind people who are helping me.
You may notice at the beginning of the topic @knro had suggested that I make that type of request, but I did not understand how to do it and what to write to be taken into consideration, I would not want to make a mistake once again in compiling the new driver request.... if I don't misunderstood it could be the solution condition.

knro, 2020-07-01:
I will close this and you can continue on #1191

closed 2020-07-01 by knro