indilib/indi #1190: Get a "File (null) load failed" error when loading PEC file for Celestron GPS

basilr, 2020-06-29:
**Describe the bug**
Running current INDI version 1.8.5 on RPi3, Celestron CGE PRO mount.
Can not load a saved PEC file, tried numerous versions of the PEC filename, tried locating the file in different folders and open permissions for everyone - execute, etc..

**To Reproduce**
Try loading a saved PEC file, will also get the error with the Celestron GPS driver in "Simulation" mode. Tried loading before and after doing "Seek Index". No change.

**Expected behavior**
File loads and PEC works.

**Screenshots**
NA

**Desktop (please complete the following information):**
NA

**Log Files**
Attached

Key? extract from log (doesn't look like much help)
[Load PEC file error.LOG](https://github.com/indilib/indi/files/4846927/Load.PEC.file.error.LOG)
:

DEBUG	460.253409 sec	: RES <W#>
DEBUG	460.253486 sec	: latitude 39.1631, sop W, PierSide E
DEBUG	460.401273 sec	: ISNewText name PEC_LOAD, text /home/pi/PEC_Data/2020-06-26/pecData_06:13.csv, names PEC_FILE_NAME, n 1
WARNING	460.401486 sec	: File (null) load failed
DEBUG	461.254461 sec	: CMD <e>
DEBUG	461.323253 sec	: RES <804F7200,3F814900#>

basilr, 2020-06-29:
Sorry, here's the PEC file..
[PIF8CW~9.TXT.csv.txt](https://github.com/indilib/indi/files/4847064/PIF8CW.9.TXT.csv.txt)

knro, 2020-07-01:
@Chris-Rowland Any ideas? I didn't know this was supported.

basilr, 2020-07-02:
Perhaps a couple more clues. Did more troubleshooting last night.

If I record a new PEC file I can play it back and the filename shows correctly, before the SET window in the INDI control pannel.

If I try to re-load that file (enter name in SET dialog) and press SET, it seems to work.

If I try to load a different saved PEC file, I do NOT get the (null) file load error. However it seems to send the first PEC file to the mount (see attached file "successful PEC load extract).

However the numbers sent to the mount (if I'm understanding what it's doing) don't match the PEC data in either file. For example: i 2, deltaDist 0.834566, rawPecdata 0.155004, rawData 11    IF i 2 means index 2, then the 0.824566 does not correspond to the Index 2 value in either PEC file ...? I'm attaching the 2 PEC files.

Finally after a reboot of the RPi, there is no PEC file showing in the INDI control pannel, and if I try to load the last file it was using I now get the (null) load failed error.

Hope this is of some help. 
Really appreciate adding the PEC feature !!

[sucessfull PEC load.txt](https://github.com/indilib/indi/files/4865481/sucessfull.PEC.load.txt)
[pecData_0238.csv.txt](https://github.com/indilib/indi/files/4865485/pecData_0238.csv.txt)
[c14average.csv.txt](https://github.com/indilib/indi/files/4865491/c14average.csv.txt)

Chris-Rowland, 2020-07-04:
I put PEC management into the Celestron driver a few months ago, it looks as if @basilr is the only one to try it.

The numbers in the file are the PEC offset in arc seconds, they match what Celestron use in their PECTool application and what I implement in the ASCOM driver.  They are processed in a mysterious way to get to and from the values that are read from/sent to the mount.  I've done this so that PEC data files generated in different ways are equivalent.

What's happening is that I wasn't updating the text control so even though there was text showing indi couldn't see it.  Have added some additional calls and everything seems OK now.

basilr, 2020-07-05:
Thank you so much! I'll give a test as soon as it's released.

Chris-Rowland: love your comment "They are processed in a mysterious way ..." 

Funny I believe I was the one who suggested the PEC feature in Oct. 2018 and I'm the only to try it ?? Sigh. This is such a great feature, and is Very appreciated, as well as all the time spent on the documentation !!

basilr, 2020-09-09:
Finally tested INDI v1.8.6 last night and loading/playback of previously saved PEC file works!
Many thanks !! - Really hope others use this (I feel) essential feature.

closed 2020-09-09 by basilr