indilib/indi #1191: Request new ASTROFOCUS driver for focuser

oldmar, 2020-07-01:
I are make sure there is no existing driver for my device.

Currently it works in windows with Ascom protocol, the request of the new driver is to make it work with the INDI protocol with  Raspberry PI

http://www.astropix.it/software/astrofocus/index.html
https://github.com/Anandir/indi-astrofocus

The family - Focusers

Device communication protocol is USB

Hardware used
Arduino mega 2560
Arduino Motor Shield
TRINAMIC QSH4218-51-10-049 STEPPER MOTOR, 1.8DEG, 1A, 0.49NM
Adafruit MCP9808 temperature control

[AstroFocus.zip](https://github.com/indilib/indi/files/4856223/AstroFocus.zip)

knro, 2020-07-01:
Hello, thank you. I added $5 to fund this driver development. We are testing this new way of funding drivers. You can contribute to this fund as well.

Anandir, 2020-07-15:
Hi @oldman, I'm surprised that you've spotted my little WIP driver ☺️ .
The driver it's not yet completed nor working, that's why there are no informations on how to build it on the repo itself.
I had a pretty busy period, so I had to suspend the development for the past weeks. I hope to put my hands on it pretty soon.
@knro , I have few questions regarding how to do some operations, can I use the INDI forum for it or there are other channels? The official chat is basically dead 😢 .
As soon as the code is working, at least the basic operations, I'll submit everything for official approval ☺️ .

knro, 2020-07-15:
@Anandir You can ask here or in INDI Development forum

Anandir, 2020-07-15:
> @Anandir You can ask here or in INDI Development forum

Thanks @knro ! As soon as I wrap my head around the problem again, I'll post all my questions on the forum ☺️ .

oldmar, 2020-07-21:
@Anandir sorry if I didn't reply quickly.
You will have read, I am not an expert in programming, I randomly read here "https://github.com/Anandir/indi-astrofocus" then I followed the suggestion of @bluthen and open a request for a new driver.
I hope for your patience, I apologize if I have created misunderstandings.
Thank you all