indilib/indi #1192: How to select the ISO for the Alignmnet Ekos module?

federicoparra, 2020-07-01:
Using Astroberry 2.0.2 (last) here, I can't seem to find a way to change the ISO within the Alignment module, using Gphoto driver with a Nikon D5100 DSLR.

knro, 2020-07-02:
Yeah there is no way now, but this is not an INDI issue either. You can file an issue in KStars here: https://invent.kde.org/education/kstars/-/issues

closed 2020-07-02 by knro

federicoparra, 2020-07-04:
Ok I will. BTW, my account on the INDILIB website is still not working :(

knro, 2020-07-05:
I deleted all your accounts, so register again.

federicoparra, 2020-07-12:
> I deleted all your accounts, so register again.

It doesn't work. When I fill in the first page of information (where there are both spam checks), I get a page with an error (begins with "undefined index failed..." but can't get to read the rest because it goes fast to a second page asking me for more information (gender, country, etc)). After I complete everything I get the link to activate the account. After I do, I login, and after I do login, I am still not logged in.... email f.parra.account@gmail.com