indilib/indi #1200: CCD_CAPTURE_FORMAT and CCD_TRANSFER_FORMAT

knro, 2020-07-26:
Implement the CCD_CAPTURE_FORMAT and CCD_TRANSFER_FORMAT in INDI

What are these for?

**CCD_CAPTURE_FORMAT** is about the camera NATIVE capture.. e.g. mono, RGB, bayer8, bayer16..etc.
**CCD_TRANSFER_FORMAT** is about how the data is sent to client / saved to disk
So this could be RAW (as is), FITS, XISF
then this facility would be available "free of charge" to all camera drivers. Maybe open INDI issue on this?

pludov, 2021-08-29:
For the CCD_TRANSFER_FORMAT: RAW, the driver should expose additional info, like:
  * endianess for 16 bits
  * the bayer pattern

knro, 2022-02-19:
Thanks, we already have bayer pattern property... perhaps we're missing the endianess

knro, 2022-02-28:
It's fixed in master.

closed 2022-02-28 by knro