indilib/indi #1201: I want to use INDI to control my device

AstroAir, 2020-08-02:
我想以通过调用INDI的方式来控制我的设备，但我不知道该如何使用，官方的API手册写得过于难懂，给我的开发造成了不小的压力；并且示例代码中的库也不全，需要一个一个下载。我曾尝试编译在这个项目中的indibase，但是会报错。能否提供一些技术支持，谢谢🙏。
PS:I am sorry that I can’t speak English.

knro, 2020-08-02:
Please provide more details on the device you want to control. Does it have an SDK? communication protocol?

AstroAir, 2020-08-02:
我是想写一个简单的client，控制相机拍摄，并存储图片

knro, 2020-08-02:
Check out this [client tutorial](https://indilib.org/developers/client-howto.html)

AstroAir, 2020-08-02:
我已经看了这份文档，但是我还是有一些疑问。比如说该如何存储照片，大多数厂家（包括INDI）的SDK手册在这一步都是省略的，应该是要使用astropy，但是我并不清楚具体的使用方法，能否稍作解释，谢谢🙏。

AstroAir, 2020-08-02:
@knro 我分别尝试了c++和pyindi，但是有一些问题，是不是所有c++的示例都在本存储库中的indibase中，我曾经尝试编译，但是这并不起作用，编译时显示多处报错（我使用的是树莓派）；接着我又尝试了单独编译，但还是不行。

knro, 2020-08-30:
No more details given and it seems there is a language barrier, so closing this for now. Please reopen if necessary.

closed 2020-08-30 by knro