indilib/indi #1213: Pegasus UPBV2 driver focuser temperature tab using a non-existent sensor

xthestreams, 2020-08-21:
**Describe the bug**
The Ultimate Power Box V2 has both a focuser and an environment sensor capability included.

The focuser driver tab features a temperature section (see screenshot), which is registering 0 degrees. In checking the pinout of the focuser hardware, there are no temperature sensor signal lines - all environmental data comes from the environmental sensor.

I believe the driver should be reading this value from the environmental sensor instead. 

**To Reproduce**
Buy a UPVv2 :-)
Install driver
Scratch head

**Expected behavior**
It should be reading the temperature from the environmental sensor.

**Screenshots**
Attached

**Desktop (please complete the following information):**
 - OS: Stellarmate
 - Version 1.5.3

**Log Files**
N/A
[image]

knro, 2020-12-09:
It is ready the temperature from the SDK provided by Pegasus. Not sure what you mean by environmental sensors? where?

xthestreams, 2020-12-10:
Okay - so there is to my knowledge only one temperature sensor on the UPBV2 and that is the Temp, Humidity, Pressure sensor - the is the source for the temperature on teh environment tab. I can't find any reference to, or even a source of, focuser temperature in teh API docs.

xthestreams, 2021-01-31:
perhaps @rishi-garrod has an idea here too!? :-)

closed 2021-11-24 by knro