indilib/indi #1216: Enable Paramount sessions to be automatically started up without operator intervention (concludes current workaround)

xthestreams, 2020-08-21:
**Is your feature request related to a problem? Please describe.**
Whilst the Paramount system was designed to be a completely robotic/automated system, the INDI driver does not expose critical features required to enable this behaviour in Ekos.

In summary the INDI driver should expose or enable a Connect from TSX to the mount AND a FindHome to reset to Home status using the following API calls;
sky6RASCOMTele.Connect() and 
sky6RASCOMTele.FindHome()"

**Describe the solution you'd like**
Specifically, when a Paramount is started from cold power up it must be Homed (find its homing sensors) to orient itself in space. If this is not done, the mount will not allow slews or tracking - so far so unsurprising.

In addition the mount itself cannot be controlled without first Connecting the TSX software to the mounts comms port - analogous to INDI driver connects. Critically this creates an interesting situation whereby the INDI driver needs to Connect to TSX and then itself issue a Connect command to the mount

Ideally the driver would use these API calls to automate the startup, potentially something like;
     if  sky6RASCOMTele.IsConnected() is false then
         sky6RASCOMTele.Connect() (then poll to ensure it stays connected as it can be unplugged or turned off is TSX dies)
     if  sky6RASCOMTele.IsConnected() & if sky6RASCOMTele.isParked is not true then
         then sky6RASCOMTele.FindHome() 

In addition, providing buttons that expose these two features in the control panel would be ideal. In fact a Home button for all devices that support Homing would be great in each Module Panel  

**Describe alternatives you've considered**
To date I am using the ScriptSkyX python scripts as follows;
    tsx.TSXSend("sky6RASCOMTele.Connect()")
    tsx.TSXSend("sky6RASCOMTele.FindHome()")

As part of my startup scripts in Ekos Scheduler however this solution has obvious problems, not least of which is the lack of good bounds checking to ensure the mount is in a known state by the INDI driver itself and with my terrible coding skills, at the moment there is no checking taking place to ensure the mount is not being homed needlessly if that has already been done (eg: if the mount is awake and connected).

I could of course write better code - which I will (I only got this working yesterday!)

**Additional context**
Add any other context or screenshots about the feature request here.

gschwim, 2020-08-21:
One caveat is last I looked TSX does not have a method to determine if the mount was already homed since power on. Please put in a feature request on the bisque forums and I can join in on the request (which is something I requested in the past). There should be no need to home the mount if it is already homed, but we have no way of really knowing that state reliably to my knowledge.

Any implementation of homing should be done such that you have to explicitly request the homing to be done by way of manual selection of the action (press button) or as part of your automation sequence (e.g. at sequence start). In cases where maybe it is requested as part of connecting to the mount there is danger of roof strike, etc, for those that "duck" the roof by parking before closure.

knro, 2020-08-24:
So to be on the same side, maybe add Auto Home switch to be enabled/disabled and if it is enabled, it would always find home after connect if not parked?

gschwim, 2020-08-24:
> Yes I thought the same thing but I realised that by Paramount's logic a mount can neither be tracking or parked if it hasn't been homed first, so provided it's in either state it's been homed and we can set an internal flag inside INDI saying IsHomed(True) - which doesn't/shouldnt stop us form re-homing if the scope gets confused (this happens more with my EQ8-R)
> 
> (I realise that the above logic doesn't mean that some idiot hasn't de-clutched it (or the SB equiv) and moved the thing, but for the purposes of a robotic observatory it's the least of our problems - but that's true of any mount without absolute encoders).
> 
> LMKWYT

It is possible this might work, but I don't think Bisque can confirm or deny that this will be a good way of verifying mount state.

gschwim, 2020-08-24:
> So to be on the safe side, maybe add Auto Home switch to be enabled/disabled and if it is enabled, it would always find home after connect if not parked?

TBH this just feels like asking for trouble. In fact, TSX doesn't allow for auto-homing. You have to explicitly request it by script or by TSX menu selection after connection. 

Maybe its better to expose the home feature as a button in the indi control panel similar to park/unpark such that it can be manually requested that way. Then, having a "home" function as part of the observatory start up in the Ekos scheduler might be better (e.g. open roof, unpark or home mount, uncap the telescope, requires explicit intentions to move the thing, etc) as opposed to the driver making that control happen. 

I do think it is worth asking Bisque if they can put this capability into TSX. I'm surprised it's not there already. I'll defer to @xthestreams if he wants to take this path. I'll weigh in in kind if he does.

xthestreams, 2020-08-25:
Comparing EKOS with ACP and Voyager, then for sure both of these allow you to “ask for trouble”, I suppose the critical thing would be to expose it and allow the observatory manager to make that determination. For a fully robotic setup it’s not unusual to allow the mount to be fully automate and homing (of every device, including things like auto focusers) should be available as part of a start up,sequence to get the observatory into a known or calibrated state. 

Again agree it’s not for everyone, but right now the only option is manual or scripting and with the rise of more mounts like my (alive again) EQ8-R that having Home sensors, I think we need to expose that as a part of a startup routine - I’ve even found it helpful mid sequence when my EQ8 got confused by too many synchs and lost its ability to slew. A quick purge of the mount model and a re-home and the sequence could continue. 

I’m starting to think that something like Voyager’s Drag Logic is the future of INDI and EKOS, we’ve already got the world’s best Distributed device server model, now all that’s needed is a better way to orchestrate events (my 20 years at an IT automation software vendor is showing!)

Dbus is great but i need a low code/no code solution, I’m just not that talented!

knro, 2020-08-25:
I can add a button to go home. Where is the latest documentation for TheSkyX calls?

gschwim, 2020-08-25:
> homing (of every device, including things like auto focusers, filter wheels, radio Astro gear?) should be available as part of a start up,sequence to get the observatory into a known or calibrated state.

Agreed. However, an unconfigured system should start with the safest defaults and allow the user to explicitly make the appropriate automation decisions.

@knro the bisque docs are here: [https://www.bisque.com/wp-content/scripttheskyx/](url)

Specifically you want `sky6RASCOMTele.FindHome()` for the homing button. Is it safe to assume this will be available via the usual scripting methods?

gschwim, 2020-08-25:
@knro - for inclusion as an automation item in Ekos should this be open as a ticket at [https://invent.kde.org/education/kstars/-/issues](url)?

xthestreams, 2020-08-25:
@gschwim makes sense - although I think the driver itself (and maybe the class as a whole?) would need to include the homing feature in order for EKOS to automate it. does what I am saying make sense or do I misunderstand how EKOS works?

Maybe I should also log a feature suggestion to add "canHome()", "isHomed()" and "goHome()" to the class? As mentioned I think this would be a great feature to add to the focuser and maybe the rotator classes too - my TCF-S can home itself on startup and it's a great feature to ensure the focuser is in a known state and that absolute positions are consistent from session to session. I can imagine that a driver for a Moonlite for example could be enhanced to create the same functionality (I have written a script to do that as part of my startup routine, works great!). 

Again sorry for all these bug/feature requests, but I figure if I document them, someone can always delete them - I REALLY am trying to be helpful, I hope it is....

xthestreams, 2020-08-26:
> Agreed. However, an unconfigured system should start with the safest defaults and allow the user to explicitly make the appropriate automation decisions.
>
> @knro the bisque docs are here: [...]

@gschwim  Correct! Ideally with a button to expose it in the Control Panel (but even inside INDI means I can use indi_setparam - one less thing).

NOTE: to do that the mount would also need to be connected, which requires a "sky6RASCOMTele.Connect()" command.

Per the initial submission, ideally "isConnected()" should be called as part of the main loop as it is possible for the mount to disconnect (eg: cable unplugged or as I have learned, if TheSkyX and INDI are on teh same Raspberry Pi - the Search feature in the INDI driver sometimes disconnects or confuses TheSkyX and it drops the connection - UGH!)

I'm happy to walk through/talk through my business logic if that helps - I know not many people are using Paramounts and INDI at the moment, but I think that might have something to do with some of the bugs - now that I have figured out how to automate, it's pretty good and getting better as I find workarounds for the bugs/missing features.

For extra credit, a toggle switch to enable Autohome on session/EKOS startup would be great (similar to how the NexDome driver has a switch to close the shutter if the dome is commanded to park).

Again, I think that along with "Unpark" this would be a great feature - remember that every time the mount is powered down it needs to be re-homed or it cannot/will not track or even point. To some extent this is true of the new EQ8-R - if connected to a hand controller it always asks to be homed on startup, it's just not 100% required if you parked before powering down. 

A long way of saying that Parking and Homing are intimately related at startup for remote/robotic setups that we will see more of as things like INDIHUB get traction. 

Note: if you take a look at what ITtelescope are doing with automation it's pretty incredible - 100% lights out operation, you simply give it a list of co-ordinates, exposure times and filter slots and it takes care of waking up, homing, etc the mount. I think INDI/EKOS will slowly take over that market once we have more mission critical features enabled.

closed 2020-08-28 by knro

xthestreams, 2020-09-03:
> @knro - for inclusion as an automation item in Ekos should this be open as a ticket at [https://invent.kde.org/education/kstars/](url) ?

Just checking - this is now considered a dead request? Not sure why it was closed - just so I know if I should keep my old scripts alive :-)

knro, 2020-09-03:
I added "Go home" switch now so that can be used. Is this sufficient?

xthestreams, 2020-12-09:
wow I didn't see that, i'll have to check it out

commit e342163a (2020-08-28): Add find home support.

Fixes #1216