indilib/indi #1217: Add Bandwidth field to filter slot data

xthestreams, 2020-08-21:
**Is your feature request related to a problem? Please describe.**
I would like to add (or work with someone to help me add) fields into the Filter class a feature for bandwidth, focus offset and exposure-time weighting that would be used to simplify and automate more intelligent exposures and automation.

**Describe the solution you'd like**
I am working on an extension to the camera module to enable the creation of "ideal" exposure times based on the ideas in SGP and here;
https://www.cloudynights.com/topic/573886-sub-exposure-tables-for-asi-1600-and-maybe-qhy163/

The intention is for a  "push here dummy" capability for those starting out in AP who don't even know where to begin or the science behind making the decisions.

To do that effectively ideally the bandwidth of each filter would be stored against each slot enabling the calculations to be done automatically assuming some kind of "autopilot" mode is enabled. 

Furthermore a "weighting" number against each would allow for an exposure offset either absolutely relative for each filter. If I had my "dream" we would be able to create a database of filters online that people could download which might also incorporate focus offsets too where not par focal.

MaximDL offers a similar feature.

**Describe alternatives you've considered**
Keeping it all in my head. ;-)

**Additional context**
Add any other context or screenshots about the feature request here.

knro, 2020-08-30:
I think this does not exactly belong at the INDI driver level, it probably belong to clients. By bandwidth, you mean the bandpass in angstroms or something else?

xthestreams, 2020-09-01:
It's a great question - I don't have an answer, but I also have questions. Some of the answers I realise might have already been thought of, but again the controls are spread all over the place - at least I think they might be...

Where do I define and manage filters and the related meta data eg: offsets, flat field exposure weightings, etc

The filter wheel driver contains slots that allow me to input the names of filters, but then none of the other data. 

Since writing this feature request, I noticed that the EKOS capture module I see that I can see some metrics including offset & focuser position, but it didn't seem like I could edit them.

I just remembered that Kstars has a device manager so I looked in there and sure enough there are even more settings, this time I can edit them.

The problem in my mind (there are a lot of problems in my mind!) is the data model for these objects. eg: there appears to be a definition of my ZWO HII filter my ASI EFW, which is great, but if I change that filter out and replace it with my Astrodon HII filter which is 3nm then I "lose" all the other filter settings. 

The reason for my confusion I guess is that the driver exposes (& stores in it's config file) the filter name for each wheel - which implies it's an INDI thing, when in reality, we're saying it's a client thing. Which makes sense, but doesn't make sense.

I think we can close this "feature request" but I personally feel as though it's digging into a broader question about data models.

eg: if I have a remote telescope and filter wheel and I want to do remote imaging, then I would think that I need to "import" all the meta data about that setup from the device itself. eg: the device in this case should tell me all I need to know about it - bandwidth, offsets, ideally it's transmission curve if I am doing science, manufacturer - all that stuff because as a client, I don't and CAN'T know it without asking someone - and rather than ask a person I should ask the device driver - in the same way my ZWO camera tells me its model, pixel size and resolution. 

Again, no answers, lots of questions - coming from a place of wanting to get the INDIHUB remote observatory model working really well.

TallFurryMan, 2020-09-07:
When reading the title I though "yeah, filter bandwidth as FITS attribute, configured in the INDI panel, good idea".

When reading the issue text, I was not so sure it was the intent anymore :)

Now that I read the second post, I think I understood properly at first. If the bandwidth information is a property of the device, and it could be helpful for clients, then it needs to be inside the INDI driver and distributed to clients for use. Thus the development is to be made first in INDI, then eventually suported in Ekos.

Looking up whether FITS headers exist for this data, I found the following attributes in the (old) Far Ultraviolet Spectroscopic Explorer [1] and a few others, illustrated with this example of values:
```
BANDWID =             99.99597 / [Angstroms] Bandwidth of the data              
CENTRWV =              949.998 / [Angstroms] Central wavelength of the data     
WAVEMIN =                 900. / [Angstroms] Minimum wavelength of the data     
WAVEMAX =              999.996 / [Angstroms] Maximum wavelength of the data     
```

As mentioned in [2] the problem remains if the optical train has more than one filter. I is also not possible to manage multi-frequency filters such as SHOO. However, these definitions could be used as a first step.

[1] https://archive.stsci.edu/fuse/DH_Final/FITS_File_Headers.html
[2] https://archive.stsci.edu/mug/mug_2008/wfpc2_mug2008.pdf

closed 2021-09-28 by knro