indilib/indi #1225: INDI (kstars? EKOS?) randomly "losing" configuration settings and/or preventing them being modified through control panels

xthestreams, 2020-08-23:
**Describe the bug**
I've been doing a lot of setup and testing recently trying to get my system into a stable/working state, which means a lot of configuration changes  - in this case to my Nexdome slaving parameters but also to my Pegasus UPBv2 power settings. 

NOTE: I am running kstars client on Mac locally, INDI on a Stellarmate remotely

I have noticed that the control panel state for each driver is unpredictable - sometimes it has "forgotten" all settings (yes, I have saved them), other times it's remembered something that I want forgotten but it won't let me change (eg: moving between Serial and Ethernet connection mode, or Enabling/Disabling device auto-search. 

The only workaround to date has been to go into the ".indi" directory of the device server and edit by hand, but even then the config changes are often lost.

This is clearly a problem when it comes to teh reliability/stability and in particular the "mission critical readiness" of INDI - is it a bug or a deeply hidden feature that I am not managing correctly?

**To Reproduce**
This is hard to describe because it's not entirely consistent - happy to let someone know when it's behaving repeatably and we can debug on the fly.

**Expected behavior**
All settings should be retained unless purged/defaulted by the operator. 

At no time should the control panel prevent changing settings.

**Screenshots**
I can provide a video that shows this happening at the time if this helps.

**Desktop (please complete the following information):**
 - OS: OSX client, Stellarmate Server
 - 1.5.3 on server, 3.4.3 client (on Catalina)

**Log Files**
NA

knro, 2020-08-23:
So I've heard reports of this, but I'd really would like to get one repeatable case. Have you noticed any sequence of events that would lead to this behavior? So just once we have one repeatable case then it's easy to figure out what's going on.

Perhaps I don't see this issue as often since my observatory settings are more or less consistent.

xthestreams, 2020-08-24:
I will let you know as soon as it happens and leave it in the same state and perhaps you can figure out out - it is the worst possible kind of bug in that way!

knro, 2020-08-30:
Any update on this issue?

skolarianer, 2020-09-08:
I have the same problem, but can't reproduce it reliably either. I'm using the latest astroberry on a RPi4.
I tried to improve this by adding udev-rules for e.g. the Pegasus PPB, but I'm not sure if this really helped. The PPB is currently not connected.
-> https://indilib.org/support/tutorials/157-persistent-serial-port-mapping.html

At the moment it happens for me with DSLR, dome and wheather (OpenWheatherMap) settings. DSLR and dome loses settings sometimes. OpenWheatherMap: I couldn't save changes after editing parameters last week. Today it worked again.

xthestreams, 2020-09-23:
I did send an email when it happened, I’ll whatsapp when it’s happening again, at least you’ll be able to see it, but as to why it happens, no clue. I am wondering if it’s related to ACLs?

xthestreams, 2020-10-19:
Hard to tell if this is two different bugs or manifestations of the same underlying bug. In some cases key buttons will not respond to "clicks" eg: changing the comms setting from Serial to Ethernet on a driver - the button simply wouldn't work, but the rest of the driver's buttons do. 
This also seems to happen at the same time (or at least it gets NOTICED at the same time) as when i am trying to Save and teh setting get lost and the save button itslef does not work as expected. I wonder if there's logic behind turning the buttons on that might be causing it or if it's a QT level bug. Whatever the bug is it's not isolated to a particular OS that I can see

knro, 2020-10-19:
The buttons not responding is a different issue and a confirmed bug when there is an INDI property removed from a contained and then re-added. But it's unrelated to configuration loss.

closed 2021-06-01 by knro