indilib/indi #1226: libindidriver.1.8.6

AradoSKY, 2020-08-24:
errors
indi_gpsd: symbol lookup error: indi_gpsd: undefined symbol: _ZN4INDI13DefaultDeviceD2Ev

indi_qhy_ccd: symbol lookup error: indi_qhy_ccd: undefined symbol: _ZN4INDI7CCDChip8setFrameEiiii

both symbols are not in libindidriver.so.1.8.6. 

expected 

GPSD, QHY5LII, and PoleMaster in EKOS.
**Screenshots**

**Desktop (please complete the following information):**
 - OS: Astroberry 2.0.3

I have no idea how 1.8.6 got on my system. My Software repository is the same as when it was built. 
 - Version [e.g. 1.7.6, GIT..etc]

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html
[spacecadet.txt](https://github.com/indilib/indi/files/5120009/spacecadet.txt)
[Symbols.txt](https://github.com/indilib/indi/files/5120011/Symbols.txt)
[log_21-29-35.txt](https://github.com/indilib/indi/files/5120037/log_21-29-35.txt)
[log_16-14-59.txt](https://github.com/indilib/indi/files/5120038/log_16-14-59.txt)

knro, 2020-08-24:
That's not an issue with INDI. I have it compiled right here and it works OK. I also checked in StellarMate, and in Ubuntu 18.04 and 20.04 and both QHY and GPSD work fine.

closed 2020-08-24 by knro

AradoSKY, 2020-08-24:
Hello Jasem,

Thank you for your reply. Hmm... the evidence I presented was from INDI. The drivers are INDI. 

It just so happens that the equipment are U-BLOX 7, QHYCCD QHY5LII, and PoleMaster. Each are working outside of EKOS. EKOS throws an exception for GPSD and QHYCCD. 

In the OS, GPSD is working. Under EKOS, INDI says it isn’t powered up. 

GPSMON, GPSPANEL, and CGPS work. In AstroBerry, the same EKOS profile works with INDIWEBMANAGER. What doesn’t work is EKOS. 

The symbol errors cut out the profile I have. It’s like INDI doesn’t see anything but ZWO and EQMOD.

I have reinstalled multiple times. I have recompiled multiple times. Using device manager, I get the same Symbol errors for GPSD and QHYCCD.

The source for these errors is missing in Libinidriver.so.1.8.6. The last few digits have “iiii” in the error. The single similar entry in the library has “ffff” but no “iiii”. This is what the error messages say. _ZN4INDI13DefaultDeviceD2Ev is similar to symbols  contained in the library. This is evidence I have provided. 

A while back, I had to run a command to do a low level truing up of services. I will seek that out. My hope is just a bad download. But, all roads point to this library not having two specific symbols thrown at it by EKOS. Where should this go?

knro, 2020-08-24:
I'd check /usr/local for any older INDI library installation as this could be the culprit. There is something else in your system that's causing this problem.

AradoSKY, 2020-08-25:
Hello Jasem,

I have checked /usr/local for any indication of an older INDI. At best, I had a version of AstroPI3. That was under the Astroberry profile. I blew that away. No change. 

This was purchased July 2020, RPI4 4G. It could not be a polluted INDI install. I stick to apt-get and the Raspberry PI page. I am not confident with nightly bleed yet. Astroberry might loose support if I did. 

LDD and NM are not pointing to an errant .so file. No missing symlink links are around. 

My GPSD is throwing some errors in Syslog. 

Aug 25 09:24:21 astroberry gpsd[1127]: gpsd:WARN: GPGSV field 3 value of 14 != actual count 24. 

I am not in an ideal spot for GPS. None of this is a show stopper.