indilib/indi #1228: Pegasus Ultimate Power Box driver Power Light and Automated Dew control toggles not working

xthestreams, 2020-08-26:
**Describe the bug**
The UBPv2 driver exposes toggle switches to enable the UPB's Power LED On/Off and Automated Dew control features.

Neither of these work in the current driver - in the case of the Power LED it immediately reverts back to the On setting (even if edited directly in the .xml config file) and in the case of the Automated Dew control features they appear to do nothing - ie: selecting them does not turn on dew heaters even when temperatures are at/close to the dew point.

Again this behaviour is consistent even if the config file is edited directly.
 
Both of these featured work under the native Windows control panel.

**To Reproduce**
As above - they simply don't work/were not implemented.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Can be provided on request.

**Desktop (please complete the following information):**
 - OS: OSX and/or native Stellarmate
 - Version 1.5.3

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

knro, 2020-09-27:
I just tested with UPB (v1) and they all work fine. Please attach log to know what the errors are.

xthestreams, 2020-12-09:
It's still not working - I am trying to debug it, from what I can see it might be a firmware problem as the PL:0 command always seems to return a PL:1 in the debug log...

xthestreams, 2021-01-11:
I'm going to take another look at this, but it seems like it might be a bug in the UPBv2 firmware on my (latest) unit.

rishi-garrod, 2021-01-29:
I have done some development previously and would be happy to take a look at this as I have just purchased a UPB and have run into this problem. Which source file is the driver in?

knro, 2021-01-30:
The source file here: 

https://github.com/indilib/indi/blob/master/drivers/auxiliary/pegasus_upb.cpp

rishi-garrod, 2021-01-30:
Thanks. I was a bit rusty but I found them in the end.

This morning I took a look at the Pegasus Windows app. I noticed a few things.

There was an update for both the app and firmware. I first tested with the old then reran the tests with the new.

The switch LED on/off Functionality:
In the old version of the app and firmware the switch off Led functionality did not work. The same result as from Indi. You send ‘PL:0’ and the response is always ‘PL:1’ and the light did not toggle on the UPB.
In the new version the app says this functionality is no longer supported, so nothing we can’t do except remove the option from the driver.

The Auto Dew heater:
In the old version of the app and firmware you send ‘PD:2’ and the response comes back in 2 separate packets. The first contains ‘PD:’ and the second contains ‘2’

With the new app and firmware you send ‘PD:2’ and the response comes back in a single packet ‘PD:2’ as expected. So I tested this again with Indi. The green indicator does light up but even if I save the settings the next time I start the driver the setting is not persisted.

I will do some more testing to try to get to the bottom of this.

rishi-garrod, 2021-01-30:
The error in the log is now 
"Error: DEW_A is not a member of Auto Dew (AUTO_DEW) property. "
"[DEBUG] CMD <PD:0>"
"[DEBUG] RES <PD:0>"

rishi-garrod, 2021-01-30:
I have found the problem and I have a fix. I will do some more testing tomorrow. It would be good if someone with a V1 could also test the fix. It should be OK but you never know until you test. 
I am thinking about removing the Led off option from 2.0 and 2.1. In 2.0 it is broken (also in the Pegasus Windows app) and in 2.1 they have deprecated it.
I could also implement the Dew aggressiveness introduced in 2.1?

xthestreams, 2021-01-31:
Amazing! Yes I noticed the same thing re: PL:1 in the logs, I was wondering if it was deprecated, seems it is - I DO WISH equipment vendors woudl stop putting LEDs in everyitng, it's hard enough competing with the light pollution OUTSIDE the dome, let alone inside! :-S
And yes, I think if you coudl determine how to implement a firmware version specific dew aggressiveness implementing that would be awesome - I personally stopped using auto-dew mainly because it was overheating my secondary, this seems like a better approach (still somewhat concerned about "auto" anything without some form of direct feedback loop eg: a temp sensor near the heater) but it's better than nothing.

Really appreciate you looking into this!

knro, 2021-01-31:
I have V1 of the controller and LED control works fine.

rishi-garrod, 2021-01-31:
Yes, it works fine on V1 but not on V2. In firmware 2.0 its still included but does not work (even from their own Windows app) and in 2.1 the app says it is deprecated.

xthestreams, 2021-01-31:
@knro - yep it seems to be a V2 hardware thing. It explains why I saw the big and you couldn't replicate it.

rishi-garrod, 2021-01-31:
They seem to be going through a lot of updates. I implemented the Aggressiveness and it did not work as expected so I switched to Windows to run their app to see how they did it. I got another update for the app and another firmware update for the UPB V2. Seems they needed to make a lot of improvements around the Auto Dew setup which sounds like good news. The aggressiveness settings are stored in Eprom and only read at startup so I need to make that clear in the driver.

rishi-garrod, 2021-01-31:
In the latest firmware update 2.2 they have fixed the LED On/Off after they deprecated it in 2.1! I can't keep up. Luckily I decided to try it before I started removing it for V2.
Just a few more things to check out. I get messages from time to time saying the "wrong number of  power/stepper/detailed sensor data has been received", looks like they made some changes there too. Might as well take a look before I push these changes.

rishi-garrod, 2021-01-31:
I have not fixed the "wrong number of power/stepper/detailed sensor data has been received" issues as I wanted to push the rest so we can get a version from the nightly build to test. I am a bit rusty with Git so I hope I have done everything right.

xthestreams, 2021-02-04:
Thanks for the great work on this so far, I've been amazed watching how quickly Pegasus are producing new versions of their firmware this week!

rishi-garrod, 2021-02-06:
Version 2.4 of the Firmware has been released. This fixes the communication problems I was seeing in 2.2 and 2.3.
I have the new serial command set. The pin used for the LED is now also being used by the stepper motor of the focuser which means if you switch off the LED the stepper motor will also switch off. Not sure whether to disable the function for V2 UPB or simply give a warning if it is switched off. Implemented the labels for Dew and USB ports. Pegasus will provide me a connection to a remote Rpi4 so I can check that the UPB V1 still functions correctly. It might be smarter at some point to split into 2 drivers but let's first see how much they diverge.

xthestreams, 2021-02-16:
Having been bitten by trying to switch it off/on and having heard back from Evans, it woudl appear that functionality will be deprecated moving forward, I would suggest we drop the feature for all releases pending further developments. A shame, I really LOVE devices that let me switch off LEDs - why do we need MORE light in our nice observatories when we go to so much trouble to get rid of it!

rishi-garrod, 2021-02-16:
In the latest commit I have removed the LED off functionality from V2 boxes. At least on the V2 it is a tiny LED.

xthestreams, 2021-02-16:
100% - it's just my pet hate with astro gear - we spend all this time and money to get to a dark sky and then the astro gear manufacturers put non-software controlled LEDs on EVERYTHING (Paramount, EQ6/8, ZWO camera, Optec focus controller, the shutter on the Nexdome, they're on EVERYTHING!)

closed 2021-04-17 by knro