indilib/indi #1234: Allow for choosing ISO on gPhoto in alignment module

federicoparra, 2020-09-14:
When using DSLR with gPhoto or the dedicated Nikon/Canon drivers, one can't choose ISO within the alignment module.
This forces the user to increase exposure time in order to get enough luminosity, which in turn puts high demands on the mount which needs to be professional enough and well-aligned to track correctly throughout the long exposure.

If ISO was selectable (as it is in the camera module), one could ramp up ISO and keep exposure low, so that no more than a few seconds are required to get a good enough picture for plate solving. This would allow operation with lower quality mounts and perhaps operation in AltAz as well.

I propose a simple ISO selector in the Alignment module, identical to the one in the camera module.

knro, 2020-09-27:
This is not really an INDI issue, it's Ekos. At any rate, it's resolved in Ekos.

closed 2020-09-27 by knro

federicoparra, 2020-09-27:
@knro it's solved in a new version?