indilib/indi #1236: Can't take photos with Pentax K1 and Stellarmate and Astroberry

Ucarre, 2020-09-17:
Hello, 
I'm trying since a month ago to take photos with Pentax k1. I have the ASI071 but i have dew issues so i want to change the camera tu the pentax. 

I have refill all the parameter in the Gphoto ccd driver and in the Pentax DSLR driver. The Shuttle port i fill it with /dev/ttyUSB0 and with DSUSB. And the camera don't do anything. I don't know if I have to install something more...

Sorry about my english😅. 

I dont know what more i can do....

Thanks!!

Ucarre, 2020-09-19:
Solucionado!!!!!!

knro, 2020-09-27:
Which drivers are you using? The pentax gphoto or pentax native? where are the logs?

Ucarre, 2020-09-27:
I solve the problem a week ago. Thank you very much for reply to my post. I used Pentax native driver and the camera in msc mode.

closed 2020-09-27 by knro