indilib/indi #1241: Status of sitech-work?

JeffRossMT, 2020-09-27:
Hello!

I have a SiTech Controller 1 that I would love to be able to use with INDI.  I see that there is a sitech_work branch but it hasn't been updated in 2 years.

I merged the changes in the sitech_work branch into the current indi-lib and the sitech drivers built okay.  However, the sitech device does not show up in K-Stars, for example.

A couple of questions, then.  What would be the correct way to test these new drivers with my controller?  Does K-Stars have to be built with the new indi-lib first?  I'm not a C programmer but I am a fairly competent python programmer and know my way pretty well around linux systems.

I absolutely can provide remote access to an Odroid N2 connected to the SiTech controller 1 with OpenVPN.

Thanks,
Jeff Ross

bluthen, 2020-09-27:
I believe that driver still requires a windows machine running the SiTech software. The INDI driver connects to a tcp port that SiTech software runs on the windows machine. 

I tried it several months back and was able to goto slew and sync, but had many other issues with it. I think there was still many features missing. I could not use the ekos align tools (plate solving), could not manual slew.

At our club observatory we have a 16" and 27" that use SiTech. I fiddled around for a month without much success, since it required a windows machine and SiTech software to be running at all times anyway to use the driver I just dropped the idea.

I actually used a different repo that had the driver so many this branch is different, but you still need the windows software and machine running.

INDI is seperate from kstars/ekos. It runs the commands to start INDI. You should not need a different kstars. You can manually start up the indi server and connect during development.

JeffRossMT, 2020-09-27:
Thank you Russell.  I had hopes that the telescope driver would connect 
directly to the controller via serial / usb so that indi could send 
commands to the controller possibly through the ASCII serial command 
interface specified in the SiTech manual.

I really hate the idea of having to buy a windows machine just to use my 
SiTech controller.  Today I've been trying to get windows xp home (the 
only windows I have) running in a VM on my Mac only to find out that the 
ASCOM and .NET  packages no longer support XP.

Jeff

bluthen, 2020-09-27:
Yeah a VM should work also, I did not try wine.

It is my understand that those commands are super low level and one would have to write a great deal of code to get it to work without the windows SiTech software. I think all the smarts are in the SiTech software and not the controller.

Maybe this forum thread can also help for some more background:
https://indilib.org/forum/wish-list/2499-mesu-mount-200-driver-request.html?start=24#48141

knro, 2020-09-27:
You can actually communicate directly with it, there is a command set. But it's very low-level and would require considerable amount of professional driver developer time to do it, over perhaps many weeks or months. Therefore, the only realistic way to have a driver for SiTech is either they hire someone to work on the INDI driver officially, or the community raises funds for the same purpose.

JeffRossMT, 2020-09-27:
Thank you Jasem, and thank you for all the work you've done on INDI.

Jeff

closed 2021-04-17 by knro