indilib/indi #1250: Make a driver for RSF focuser

echoman7, 2020-10-29:
[RSF_focuser_Protocol_200313_EN.pdf](https://github.com/indilib/indi/files/5457979/RSF_focuser_Protocol_200313_EN.pdf)
[RSF_focuser_Protocol_200313_KOR.pdf](https://github.com/indilib/indi/files/5457980/RSF_focuser_Protocol_200313_KOR.pdf)

echoman7, 2020-11-24:
I will wait for it to be made.

boulandab, 2020-12-30:
Rainbow Astro RSF driver documentation https://indilib.org/index.php?option=com_mtree&task=viewlink&link_id=146

If you can please test it and report back if you get any issues.

echoman7, 2021-01-05:
Thank you very much.

echoman7, 2021-01-12:
A bug has been found in "Rainbow Astro RSF driver" 
We will inform you of the bug part with pictures. 
[image]

boulandab, 2021-01-12:
Can you clarify what you mean? The desired steps on the right are go to and not relative. So putting in 3000 should take you to 3000. The range is from 0 to 16000.

echoman7, 2021-01-12:
When the first part A is 8000, if you put 1000 in B, it should be 7000 in A, which is 7999.
  
[image]

closed 2021-01-12 by echoman7

echoman7, 2021-01-13:
When the first part A is 8000, 
if you put 1000 in B, it should be 7000 in A, which is 7999.

knro, 2021-01-13:
A is CURRENT position
B is TARGET position

After you set B then click the "play" icon next to it, the focuser should go there so that A equals B eventually. If this is not happening then please attach the focuser logs.