indilib/indi #1252: Apogee driver contains embedded $PATH to /home/jassem and files to start

xthestreams, 2020-11-05:
**Describe the bug**
When starting the Apogee driver I get the following error:
2020-11-05T02:46:19: [ERROR] Error finding USB device: libapogee:/home/jasem/Packages/libapogee3-package-stable/libapogee3/linux/GenOneLinuxUSB.cpp(72):failed to claim usb interface:0 
2020-11-05T02:46:19: [INFO] Searching for Apogee CCD... 
2020-11-05T02:46:03: [ERROR] Error opening camera: libapogee:/home/jasem/Packages/libapogee3-package-stable/libapogee3/linux/GenOneLinuxUSB.cpp(476):UsbRequestIn failed with error -9. RequestCode = c2 : Index = 0 : Value = 2:1 

**To Reproduce**
Clean install/update
Plug in USB Alta camera
Select and connect driver
See error appear

**Expected behavior**
It should connect to the camera.

**Screenshots**
2020-11-05T02:46:19: [ERROR] Error finding USB device: libapogee:/home/jasem/Packages/libapogee3-package-stable/libapogee3/linux/GenOneLinuxUSB.cpp(72):failed to claim usb interface:0 
2020-11-05T02:46:19: [INFO] Searching for Apogee CCD... 
2020-11-05T02:46:03: [ERROR] Error opening camera: libapogee:/home/jasem/Packages/libapogee3-package-stable/libapogee3/linux/GenOneLinuxUSB.cpp(476):UsbRequestIn failed with error -9. RequestCode = c2 : Index = 0 : Value = 2:1 

**Desktop (please complete the following information):**
 - OS: Stellarmate 
 - Version 1.5.1

**Log Files**
2020-11-05T02:46:19: [ERROR] Error finding USB device: libapogee:/home/jasem/Packages/libapogee3-package-stable/libapogee3/linux/GenOneLinuxUSB.cpp(72):failed to claim usb interface:0 
2020-11-05T02:46:19: [INFO] Searching for Apogee CCD... 
2020-11-05T02:46:03: [ERROR] Error opening camera: libapogee:/home/jasem/Packages/libapogee3-package-stable/libapogee3/linux/GenOneLinuxUSB.cpp(476):UsbRequestIn failed with error -9. RequestCode = c2 : Index = 0 : Value = 2:1

xthestreams, 2020-11-07:
Good news - the bug disappeared when I upgraded the firmware from V23 to v33 - Jassem, if you agree I'll quickly edit the web page to reflect this as a "known bug"?

knro, 2020-11-07:
Thanks for the update. Also, I replaced the path with base file instead of full path.

closed 2020-11-07 by knro

xthestreams, 2020-11-08:
I've added a review in rather than change the main page until we talk about the best way to do it.