indilib/indi #1259: I want to write an indi C + + client

AstroAir, 2020-11-14:
Hello, long time no see.After a period of study, I have started to write my own program.Because of the good device support of indi, I want to write a client that can control more devices with indi.I've read indi's API manual, but I still have some problems.I have tried to compile the example, but it is not successful. I don't know what went wrong. I hope you can provide some technical support. Thank you！

TallFurryMan, 2020-11-14:
You can pick examples from how Ekos is doing it at https://invent.kde.org/education/kstars, check the INDI folder and the client written for Qt. If you have issues with the example, we'd be interested in the failure log.

AstroAir, 2020-11-14:
@TallFurryMan Thank you for your reply.I have downloaded the source code of KSTAR and have a preliminary look at it.The code has very detailed comments, which is convenient for me to read.But it seems that this is not very suitable for me. I want to write a server, and all the results are displayed on the web page, so QT may not be used.In addition, I will have a problem, many software will have one indiclient.cpp But there are indi programs. Why? Why split into two?

knro, 2020-11-17:
What's not successful exactly in compiling the client?

AstroAir, 2020-11-26:
@knro Sorry,I have just finished my job and didn’t do clients.I’ve tried the example ,but when I tried to connect with the indi server,it didn’t work at all.I’ve a very big problem with the steps to connect the server.Which programs should be run at first and then .And the README is not very clear for the green hand just like me.Could you answer my questions?Thanks.

d33psky, 2020-12-30:
@AstroAir are you still stuck on the example where you cannot connect a client to an indiserver ?

AstroAir, 2021-01-02:
@d33psky  Yes, I'm still studying how to communicate with indi recently, but I don't have any clue. Can you provide an executable example to facilitate my understanding

d33psky, 2021-01-03:
Maybe start here ?
In 1 terminal : 
```
indiserver -v indi_simulator_telescope
```
In a 2nd terminal : 
```
indi_getprop
```
And see that the indiserver logs the new arrival from indi_getprop on 127.0.0.1.

Now make a change, connect the simulator telescope : 
```
indi_setprop 'Telescope Simulator.CONNECTION.CONNECT=On'
```
and note with subsequent
```
indi_getprop 'Telescope Simulator.EQUATORIAL_EOD_COORD.*'
```
calls that the RA coordinate changes over time.

AstroAir, 2021-01-03:
@d33psky Thanks.I have tried this method, it can be used, but I hope to write a C + + client, can you help me？

d33psky, 2021-01-03:
For that I simply refer you to https://indilib.org/developers/client-howto.html

closed 2021-04-17 by knro